pbatard/rufus #2688: In-place upgrade to 24H4 - Setup.exe doesn't load.

lectronix, 2025-02-11:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ x] Running Rufus on a different computer.
- [ x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
The setup.exe created in the root of the USB doesn't open. It briefly shows a black window then closes. No errors or anything. I tried changing different options in Rufus (BIOS,UEFI,GPT,MBR & customize Windows options except the hardware option obviously. Tried three different 24H2 images (x64,US&UK).

I've had to chop out part of the 'Extracting....' bit of the log as it was too long to fit 65536 characters on this page.

Please advise! Thanks!

Log
---
Rufus x64 v4.6.2208
Windows version: Windows 11 Pro x64 (Build 22000.2538)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found 2351 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- MS/MS-Pro USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found UAS (USB 3.0) device 'ASMT ASMT105x UAS Device' (174C:55AA)
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'Generic- SM/xD-Picture USB Device' (058F:6364)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3892, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {46305B5E-9043-4694-A216-1E30A0F547C7}
Max parts: 128, Start Offset: 17408, Usable = 32015645184 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {11D3221D-2EE2-4E28-853A-C02F36857C13}
  Size: 29.8 GB (32013529088 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {398CC2DB-C088-44CB-8B69-27888B0DED1B}
  Size: 1 MB (1064448 bytes)
  Start Sector: 62528508, Attributes: 0x9000000000000000
Found 5 additional UEFI revocation filters from remote SBAT
[... 710 lines cut ...]
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3824, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {B543EECC-6450-4CB8-8BB7-23100CEE4B05}
Max parts: 128, Start Offset: 17408, Usable = 31457246208 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {6E6B0712-1FB7-4AD9-8ECF-33F725FB325E}
  Size: 29.3 GB (31455113216 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {08FB995E-F3B7-48A3-8061-6CCE255C38A2}
  Size: 1 MB (1064448 bytes)
  Start Sector: 61437852, Attributes: 0x9000000000000000

Computing hash for 'C:\Users\LECTRONIX-SERVER\Desktop\Win11_24H2_English_US_x64.iso'...
  MD5:    bb2acb101d4a360c71530e4be00ec43c
  SHA1:   a81079ba4b9a9ce29a7a2a20cd34266ac21becce
  SHA256: b56b911bf18a2ceaeb3904d87e7c770bdf92d3099599d61ac2497b91bf190b11

pbatard, 2025-02-11:
Hmmm, you seem to be using Windows 21H2 (Build 22000.2538) which is pretty old.

It's likely going to take a while for me to test this to see if I can replicate your issue.

lectronix, 2025-02-11:
Thanks! Yeah it is. To be honest I didn't try Rufus on another machine because I've used it for years on this one without issue. It's always worked fine. I can go and get a laptop and try Rufus if you think it'll be worth it. I think I have one with 22H2 on it.

pbatard, 2025-02-13:
I installed Windows 11 21H2 (on a real physical machine that doesn't have TPM), made sure that it got fully updated through Windows Update. Then I created a Windows 11 24H2 USB drive using Rufus 4.6 and I was able to run `setup.exe` and proceed with an in-place upgrade without trouble.

Therefore I'm afraid that I have to conclude that your issue is purely environmental and that I will not investigate it any further.

closed 2025-02-13 by pbatard