pbatard/rufus #2689: Issue with TPM bypass when creating image

slloyd-skl688, 2025-02-12:
Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.

I am including a log 

- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

I have, I found 2623 but I don't think my temp folder is symlinked  I found 2574 but I am using rufus 4.6.2208 so that is not the issue. 

I found this video where they are having the same issue: 

https://www.youtube.com/watch?v=Sd1bQaR605k

But the fix in that video (run as administrator) did not resolve the issue. 

- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.

Yes. 

- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Yes

Additionally (if applicable):
- [ ]  I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.

Yes, for both USB drives that I tried. These are both brand new, out of the box, USB drives 

- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
Yes
  - [ ] Plugging the USB into a different port.
 Yes

- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

I think so, I am not sure if the hashs are included,  but I tried with both the image downloaded via rufus as well as the official download from Microsoft and the error applied to both images

Issue description
-----------------
I am attempting to make a windows 11 ISO without the TPM check. I set the Partition Scheme to be "MBR" and check the "bypass tpm check" checkbox. The ISO appears to run right until then end and then I get the error. Any assistance you can provide would be greatly appreciated. 

Log
---
```
Rufus x64 v4.6.2208 (Portable)
Windows version: Windows 11 Pro x64 (Build 22621.4317)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring volume '\\?\Volume{0a08d05f-9beb-11ed-9e1e-5c60ba401961}' because it has more than one extent (RAID?)...
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)

slloyd-skl688, 2025-02-12:
1 device found
Ignoring volume '\\?\Volume{0a08d05f-9beb-11ed-9e1e-5c60ba401961}' because it has more than one extent (RAID?)...
Disk type: Removable, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 30155, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x051B98CF
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 231.0 GB (248033284096 bytes)
  Start Sector: 2048, Boot: Yes
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Found 5 additional UEFI revocation filters from remote SBAT
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 26100.1742)
  Size: 5.4 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_24H2_English_x64.iso (5.4 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
Timeout while retrieving conflicting process list

Format operation started
Requesting disk access...
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Could not open GUID volume '\\?\Volume{2e1ce26c-e8ce-11ef-9ead-806e6f6e6963}': [0x00000021] The process cannot access the file because another process has locked a portion of the file.
Ignoring volume '\\?\Volume{0a08d05f-9beb-11ed-9e1e-5c60ba401961}' because it has more than one extent (RAID?)...
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 231.0 GB)
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive3 for exclusive write access
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume \\?\Volume{2e1ce26c-e8ce-11ef-9ead-806e6f6e6963}\
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Disabling file indexing...
Ignoring volume '\\?\Volume{0a08d05f-9beb-11ed-9e1e-5c60ba401961}' because it has more than one extent (RAID?)...
Opened \\?\Volume{2e1ce26c-e8ce-11ef-9ead-806e6f6e6963} for exclusive write access
Writing Partition Boot Record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{2e1ce26c-e8ce-11ef-9ead-806e6f6e6963}\ as E:
Extracting files...
[... 2064 lines cut ...]
Created 'E:\setup.exe' bypass wrapper (from embedded)
Mounting 'E:\sources\boot.wim[2]'...
Could not mount 'E:\sources\boot.wim [2]' on 'C:\Users\slloyd\AppData\Local\Temp\RufEC64.tmp': [0x00000522] A required privilege is not held by the client.
Finalizing, please wait...
E:autorun.inf already exists - keeping it
NTFS Fixup (Checkdisk)...

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
1 device found
Ignoring volume '\\?\Volume{0a08d05f-9beb-11ed-9e1e-5c60ba401961}' because it has more than one extent (RAID?)...
Disk type: Removable, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 30155, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03C31EE6
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 231.0 GB (248033284096 bytes)
  Start Sector: 2048, Boot: Yes
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)

```

slloyd-skl688, 2025-02-12:
Here are my checksums:

MDA5: bb2acb101d4a360c71530e4be00ec43c

SHA1: a81079ba4b9a9ce29a7a2a20cd34266ac21becce

pbatard, 2025-02-12:
```
Could not mount 'E:\sources\boot.wim [2]' on 'C:\Users\slloyd\AppData\Local\Temp\RufEC64.tmp': [0x00000522] A required privilege is not held by the client.
```

Well, that's pretty explicit. Your Windows is configured in such a way that it doesn't allow Rufus to mount WIM images, which any **vanilla** (i.e. unmodified) standard installation of Windows should be able to do.

Which means that your problem is purely environmental, and it will be up to you to figure out how exactly your installation of Windows was misconfigured to prevent a standard operation from working.

This is not something I will be able to fix in Rufus, as anybody can remove components or restrict their Windows configuration in such a manner that breaks Rufus, and the expectation, if you want to be able to use Rufus, is that your Windows installation is standard and not restricted enough to allow it to work.

In other words, and especially since I haven't gotten any other reports for this specific error, the problem is on your side, and you will have to find out what it is. And while I will close this issue as a result, you will still be able to post replies, so if you do figure out how you misconfigured Windows in such a manner that it prevents WIM images from being mounted, please don't hesitate to follow up here, so that it might help other people.

closed 2025-02-12 by pbatard

pbatard, 2025-02-12:
Rufus runs with elevated privileges, so it would run as the Administrator user.