pbatard/rufus #2691: cannot create bootable ARM64 windows setup usb

limulidae, 2025-02-14:
Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Recently I tried to create setup usb for Windows ARM64.
I found the following issues:
(1) the file system is ntfs only, which is not supported yet. I have to manually change to fat32. 
fat32 is supported in my laptops. my laptops don't support ntfs.
Rufus needs to force fat32. (I dont have fat32 option in rufus)
(2) the install.wim is larger than fat32 supported size. 
If Rufus can help to split install.wim into multiple install.swm, that would be great.

Log
---
```
<FULL LOG>
```

pbatard, 2025-02-14:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah

And **yes** that applies for ARM64. NTFS is 100% useable with UEFI on ARM64. There's **nothing** in the UEFI specs that forces the use of FAT32, and Rufus uses [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) (which is compatible with Secure Boot) to ensure that, yes, your ARM64 machine will be able to boot from NTFS regardless of whether your UEFI firmware has an optional native NTFS driver (besides the mandatory FAT32 one) or not.

Please don't assume that an application will have a problem, on account of some half-truth and misinterpretations that you get from the Internet. Instead try, and, if that doesn't work, report on the *actual* errors you are experiencing.

closed 2025-02-14 by pbatard

limulidae, 2025-02-14:
This is the actual errors I am experiencing.

I download windows arm64 iso -> use rufus to build setup usb -> surface pro 9 5g boot from usb, failed.

pbatard, 2025-02-14:
That's a Surface Pro boot issue, not a Rufus bootable drive creation issue. I can assure you that the drive created by Rufus from Windows 11 ARM64 does work for the installation of Windows on ARM64 based UEFI compatible devices.

Also, since your issue happens post media creation, and per the check list, you were supposed to provide the full log from Rufus of how you created the drive, so that I can validate that there is no potential hiccup there.

Most likely, you have you Secure Boot set to _Microsoft only_ instead of _Microsoft and third party_ per #2137.