pbatard/rufus #2692: Setup.exe does not open. A console window flashes for a brief second, but then closes. Rufus 4.6.2208 (Portable) - Win11_24H2_English_x64

weiluntong, 2025-02-15:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
This is actually a revisit of an issue others have cited at these two places:

- A comment by by @ultradj83 in [#2568](https://github.com/pbatard/rufus/issues/2568#issuecomment-2495951244)
- Issue #2688 by @lectronix

I'm also experiencing an issue in which the `setup.exe` just does not load. A console window opens for a second before closing. I've tried running `setup.exe` from Powershell to see if it would maybe capture any output from that console window, however, it just opens a new console window and again closes without displaying anything. At the advice of another issue (though it seems I can't find that comment), I've also monitored Windows Event Viewer, but have not found any new events created when I ran `setup.exe`.

Of interest, this issue only occurs when I select one of the options for installation provided by Rufus. The only option I select is, "Remove requirement for 4GB+ RAM, Secure Boot and TPM 2.0." If I do not select that option and create an installation media, the `setup.exe` runs on the computer I'm trying to upgrade.

I think of note is that the log will reflect where the media was created, but not the computer in which I am attempting to upgrade Windows. The computer I am trying to perform an in-place installation on is on Windows version 22H2. Which is about a year newer than the one mentioned in the issue listed above, so I don't *think* it's an issue with the Windows version?

Log
---
[rufus.log](https://github.com/user-attachments/files/18808746/rufus.log)

weiluntong, 2025-02-15:
Workaround
---------
I actually found a workaround too that others may be interested in. The computer I'm trying to install too already had to bypass the requirements once when I was upgrading from Windows 10 Pro. Interestingly enough, I was able to run `setup.exe` to upgrade to 24H2 by creating an installation media without applying any modifications to the Windows install media.

pbatard, 2025-02-15:
There's only so many ways I can repeat that these issues are purely environmental. I have not been able to ever replicate the problem on  a **vanilla** (read unmodified) and fully up to date installation of Windows.

So this means that you either have added or removed an extra component to/from your installation that is preventing `setup.exe` from running.

At this stage, I don't believe it is an antivirus issue since neither [here](https://www.virustotal.com/gui/file/a74dbfc0e2a5b2e3fd4ad3f9fdaea0060c5d29a16151b9a570a9bd653db966a7/detection) and [there](https://www.virustotal.com/gui/file/629fdda27e200ec98703a7606ca4e2d0e44c578341779e4d5c447d45fc860c69/detection) report any false positives.

Now, considering that I just cannot replicate that issue, I'm afraid it is up to **YOU** to find out how exactly your Windows installation deviates from up to date vanilla Windows, and report on your findings, because I'm not going to spend my time installing Windows over and over again, and fiddling with various settings or adding/removing components, in the hope that I *might* be able to replicate the problem. As far as I can tell, the problem is with your own customisation of the Windows platform you are trying to run `setup.exe` from, so you **will** need to investigate and find how you (mis)configured Windows in a manner that prevents a [very simple executable](https://github.com/pbatard/rufus/blob/master/res/setup/setup.c) that doesn't do anything even remotely fancy from running.

And yes, as a result, I will continue to close such issues, especially as you chose not to provide a log, which prevents people from finding potential clues with regards to the platform you are upgrading from and how the media was created.

closed 2025-02-15 by pbatard