pbatard/rufus #2693: Feature Request: Add button for popup with Power keys/Cheat modes

Ralf-AM, 2025-02-15:
Can you maybe add a button e.g. to the right of the "Log"-button that opens a popup window with all the Power keys/Cheat modes as described at the end of the FAQ - there are too many keyboard shortcuts to remember them all. 
For now, a link to the corresponding section of the FAQ would be enough (which could certainly be implemented quickly), but since internet is not always available everywhere, a direct pop-up window with the buttons and explanations would be much better.

pbatard, 2025-02-15:
This has been asked before, but I am not planning to do anything like that on account that I don't want to present the cheat modes as something that is officially endorsed by the application. I genuinely don't want people to find these unless they have a clue what they are doing, and pointing everybody and their grandmother to them, and making them look as if they are meant to be as permanent a fixture as the other UI options (which they aren't) is not something that I see as beneficial for Rufus users.

As such, I will have to reject this request.

closed 2025-02-15 by pbatard

Ralf-AM, 2025-02-16:
OK, I understand your arguments, but would it be possible to add a freely assignable button so that one can specify a URL, a script or a program in the settings that is executed when you click the button. You could use the URL of the Rufus homepage as a default and as an explanation you could state that users can use it to create quick access to a program that is important to them (e.g. starting a backup program to save the contents of the USB stick before it is formatted, etc.) 
[And if a user knows what he's doing, he can enter the URL for the keyboard shortcuts - or create an own help file for Rufus and link it to the freely assignable button]

pbatard, 2025-02-16:
>  would it be possible to add a freely assignable button

Too custom. Too specific for **your unique workflow**. See https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.