pbatard/rufus #2697: Fido and download Isos Windows 10/11

seba2282, 2025-02-20:
Fido downloader have updates sources to download updated ISOs for Rufus don't tell later about how bootloader EFI is revoked?
I see much that monits after download iso from Microsoft site /Media Creation Tool or when download from UUP Dump with newest patches. With Windows 10 22 H2 or 11 23 H2 are that situations generally most.

pbatard, 2025-02-20:
Unlike UUP, which **creates** custom unofficial ISOs from bits and pieces they obtain from the Microsoft binary servers, Rufus downloads **retail** ISO directly from Microsoft, so, as long as Microsoft does not provide updated **retail** ISOs, Rufus cannot provide Windows 10 versions with updated bootloaders.

If you are unhappy about this situation, please complain to Microsoft so that they update their **retail** Windows 10 ISOs, because, again, all Fido does is pick these ISOs directly from Microsoft servers. It's not because anybody can create **unofficial** ISOs where the revoked bootloader warnings are removed that I want to do so in Rufus, especially as using **retails** Windows ISOs is the only way to make sure that somebody has not maliciously tampered with your Windows installation data: https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_can_I_validate_that_a_Windows_ISO_is_a_genuine_retail_version.

Also, with Windows 10 reaching End Of Life soon, I really have no plans to add workarounds in Rufus to sort out the revoked bootloader situation there (especially since, again, it is up to Microsoft to fix the situation in their retails ISOs, and not Rufus).

closed 2025-02-20 by pbatard

seba2282, 2025-02-20:
@pbatard General all ISOs from Microsoft MCT or if you download updated ISO from Microsoft site, are flagged. Without 11 24H2 because that was after all putted and have only few months. 
And I understand, is easy closed topic. But Rufus should have option to download updated bootloader for fix? I make you to remember but fix came with updates. From UUP Dump is option integrated them. So de facto newest compilation iso shouldn't be flagged.

pbatard, 2025-02-20:
> But Rufus should have option to download updated bootloader for fix?

It's not that easy, because you need to update the .wim's as well. I investigated this in https://github.com/pbatard/rufus/issues/2244#issuecomment-2400721366 and found that it was just not realistic for Rufus to properly update all the bootloaders that might have been revoked.

Again, if you are unhappy about the Windows 10 situation with revoked bootloaders, don't complain to me. Complain to Microsoft. It's **their** job top update their images so that they don't use revoked bootloaders and people shouldn't have to patch or recreate ISOs to avoid this. As long as **you** don't complain to the relevant people, the situation will not change, and Microsoft will continue to feel entitled to do a bad job with not updating ISOs that are known to contain vulnerable bootloaders, even as the OS has not yet reached EOL.

seba2282, 2025-02-20:
I can't understand one. Why to verification point is site adguard? They have genui ISOs? Because you prefer that Russian? Can't give link to some iso verifer?

pbatard, 2025-02-20:
> Why to verification point is site adguard?

Because Microsoft *used to make* the SHA-1 of their retail ISOs publicly accessible in a convenient list on MSDN, but about 3-4 years ago, they put that list behind a subscriber access.

So we had to find an alternative site that provides these same SHA-1's (the SHA-1's for the retail Microsoft ISOs are **unique** &mdash; that's the whole point), and it happens that adguard does, and it has a convenient search function.

Oh, and your ISO verifier is Rufus. You just need to compare to locate and compare the SHA-1. And no, I can't go around maintaining a list of SHA-1's myself. That's why I delegate this to adguard. And trust me, if adguard ever start to publish SHA-1's that don't match the ones from the actual retail ISOs, it will be known (because it will be super obvious). So, yeah, a site that publishs SHA-1 that **everybody else** can easily validate for accuracy can be trusted, regardless of where they reside.