pbatard/rufus #2699: Business Proposal, Improved RUFUS github Titile

OmarBadr145, 2025-02-22:
Dear Sir / Madam, 

I am writing in connection with a fantastic business proposal. The proposal is simply to change the title of the "README" tab on the RUFUS github repository to include the word "Service" in order to complete the RUFUS acronym.

The new proposed title would be: "Rufus: The Reliable USB Formatting Utility Service"

Kind Regards
Omar Badr

pbatard, 2025-02-22:
Rufus is not a service. A service is an application that runs permanently in the background. If I were to change the 'S' to 'Service' it would give the wrong impression that Rufus will install an executable to run in the background, which it does not, and which would worry users who understand what the common definition of a service is.

&rarr; Rejected.

closed 2025-02-22 by pbatard

OmarBadr145, 2025-02-22:
But it would be so sick tho???

pbatard, 2025-02-22:
No it wouldn't. Accuracy is what's "sick". Calling an orange an apple isn't "sick"