pbatard/rufus #2701: Some thoughts about DLL planting attacks prevention

dimhotepus, 2025-03-01:
May be you find it suitable: 

* https://learn.microsoft.com/en-us/cpp/build/reference/dependentloadflag

> When the operating system resolves the statically linked imports of a module, it uses the [default search order](https://learn.microsoft.com/en-us/windows/win32/dlls/dynamic-link-library-search-order). Use the /DEPENDENTLOADFLAG option to specify a load_flags value that changes the search path used to resolve these imports.

> In both cases, if you specify the link option /DEPENDENTLOADFLAG:0x800 (the value of the flag LOAD_LIBRARY_SEARCH_SYSTEM32), then the module search path is limited to the %windows%\system32 directory. It offers some protection from planting attacks on the other directories. For more information, see [Dynamic-Link Library Security](https://learn.microsoft.com/en-us/windows/win32/dlls/dynamic-link-library-security).

* If you want to load only signed DLLs, you can also try:

https://learn.microsoft.com/en-us/cpp/build/reference/integritycheck-require-signature-check

pbatard, 2025-03-01:
Thanks, I appreciate the tips, but unfortunately, I need something that works with both MSVC **and** MinGW/gcc (the main releases are built using MInGW/gcc and as a matter of fact, the last DLL side-loading issue did not affect the MSVC builds of Rufus), so these MSVC specific build flags can't be applied as is there. From what I can see there is simply no equivalent for MinGW (and yes, I tried `-Xlinker /DEPENDENTLOADFLAG:0x800` but this is for make files that still invoke MSVC linker and not `ld`).

Now, if there is some way that `/DEPENDENTLOADFLAG:0x800` actually translates to a bit being toggled in the resulting executable, I'll see if I can use that, but my current guess is that this translates into adding/generating extra code.

At any rate, and this is for any MS people listening, `/DEPENDENTLOADFLAG:0x800` should not be a linker option (though it is of course better than nothing) but a manifest option because declaring how DLL loading should behave is exactly the kind of thing a manifest, which is intended to be processed by the core system loading both the initial executable and its DLL dependencies, is meant to do and it absolutely baffles me that Microsoft **still** has not created a manifest option that can take care of disabling side-loading for executables that do not need to load DLLs from anywhere else but the system ever. At this stage, it is almost criminal for Microsoft not to have done so, and instead wasting their efforts in limited MSVC build options, as one could just re-manifest existing executable to easily sort existing side-loading issue... Still, hopefully, `/DEPENDENTLOADFLAG:0x800` might translate to something that could be applied to a MinGW/gcc build, so thanks for pointing that out.

Wack0, 2025-03-03:
apparently this goes in the PE load config directory (this structure has DependentLoadFlags element)

pbatard, 2025-03-03:
Yeah, this is indeed documented at https://learn.microsoft.com/en-us/windows/win32/debug/pe-format#the-load-configuration-structure-image-only and from what I can see, the only differences between an executable where the option is set and one without is the `DependentLoadFlags` field and the various checksums (of the latter there appear to be 5 instances: 4 PE ones and an SHA-256).

Now, the "fun" part is that MinGW's `ld` does not create a `load-configuration` PE directory entry by default, so, even if we disregard the checksums, we can't just edit the binary and flip a bit there. Instead, it looks we'd have to go through a whole shebang of creating a custom linker script to insert the directory entry.

At this stage, since nobody else on GitHub appears to have done something like this, I'm trying to wrangle ChatGPT into producing something I can use, but, unsurprisingly, after 1 hours of trial and error, I'm getting absolutely nowhere...

pbatard, 2025-03-03:
Well, ChatGPT is NO_GO, and I don't have days to invest empirically learning the arcane syntax of GNU `ld` so that I can coax a PE directory entry into existence (and yes, I know MinGW's LLVM does have the option I am after through a flag, but I want to be GNU through and through). So, while I will add the `/DEPENDENTLOADFLAG:0x800` flag to the MSVC builds, I'm just going to continue trying to keep MinGW happy with delay loading.

closed 2025-03-03 by pbatard

pbatard, 2025-05-13:
Some further testing with Visual Studio confirms that `/DEPENDENTLOADFLAG:0x800` only applies to first party DLLs, but not dependent DLLs, which makes it quite limited in scope.

For instance, even if you have `/DEPENDENTLOADFLAG:0x800` but have some code that hooks into `setupapi.dll` or `crypt32.dll`, then you may find (as is the case with current Rufus) that MSCV still issues some side-loaded calls to dependent DLLs `cfgmgr32.DLL` (for `setupapi.dll`) or `CRYPTBASE.DLL` (for `crypt32.dll`). Which means you still have to explicitly tell MSVC to delay-load `setupapi.dll` and `crypt32.dll` if you don't want to be vulnerable to `cfgmgr32.DLL` or `CRYPTBASE.DLL` side-loading...

So, yeah, that `/DEPENDENTLOADFLAG:0x800` is a stopgap solution at best since you may still be subjected to side loading from Windows' system DLLs that **don't** have any `Dependent Load Flag` set:

```
C:\Projects\rufus\src>dumpbin /LOADCONFIG C:\Windows\System32\cfgmgr32.dll
Microsoft (R) COFF/PE Dumper Version 14.38.33133.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file C:\Windows\System32\cfgmgr32.dll

File Type: DLL

  Section contains the following load config:

            00000140 size
                   0 time date stamp
                0.00 Version
                   0 GlobalFlags Clear
                   0 GlobalFlags Set
                   0 Critical Section Default Timeout
                   0 Decommit Free Block Threshold
                   0 Decommit Total Free Threshold
    0000000000000000 Lock Prefix Table
                   0 Maximum Allocation Size
                   0 Virtual Memory Threshold
                   0 Process Affinity Mask
                   0 Process Heap Flags
                   0 CSD Version
                0000 Dependent Load Flag
    0000000000000000 Edit List
    000000018004E1C0 Security Cookie
                  (...)
```

And of course, the same is true for `crypt32.dll`. I guess that also explains why I got the `CRYPTBASE.DLL` local load with MinGW, as I have not been delay-loading `crypt32.dll` there (since, because it's a pain to add, I only delay-load DLLs where I see issues, and I had never seen `crypt32.dll` being side loaded), whereas I do blanket delay-load all the DLLs for MSVC (since there, it's just a matter of adding the DLL name in the project conf).

I guess Microsoft was more worried about breaking legacy apps from some of their corporate customers, that might *foolishly* rely on side-loading local DLLs from system DLLs, than they are about security, and decided **not** to set the `Dependent Load Flag` to `LOAD_LIBRARY_SEARCH_SYSTEM32` for their system libraries...