pbatard/rufus #2702: Is there a way to disable UEFI revocation checks? (or, "Timeout while retrieving conflicting process list" is eating up my time)

Qix-, 2025-03-04:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it. (note: My log doesn't have that for some reason; this is a full log after a clear)

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm building a bootable USB drive on windows for an OS under development. I'm doing this frequently at the moment (every other minute or so) as I'm trying to test a bug on real hardware. Rufus keeps trying to check the UEFI revocation status but it's completely irrelevant for my case, and it seems that there's something that makes it hang over and over again, chipping away at time (upwards of 5-10 seconds each time I need to flash it). According to the logs, it seems to time out looking for another process.

Is there a way to disable this step entirely?

Log
---
```
Timeout while retrieving conflicting process list

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive4 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 58.4 GB)
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{e91a2f06-f869-11ef-86fd-d45d64d7cce9} for exclusive write access
Size : 58.4 GB 122550144 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 1802:3e49
806 Reserved sectors, 14957 sectors per FAT, 2 FATs
1914366 Total clusters
1914365 Free clusters
Clearing out 30784 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 806
FAT #1 sector at address: 15763
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting label...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{e91a2f06-f869-11ef-86fd-d45d64d7cce9}\
Successfully remounted \\?\Volume{e91a2f06-f869-11ef-86fd-d45d64d7cce9}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: E:\[BOOT]\0-Boot-NoEmul.img (2 KB)
Extracting: E:\[BOOT]\1-Boot-NoEmul.img (2.8 MB)
Extracting: E:\EFI\BOOT\BOOTIA32.EFI (244 KB)
Extracting: E:\EFI\BOOT\BOOTX64.EFI (244 KB)
Extracting: E:\M-6TD9M6QGRUGMZEH3HCRMGW0D1 (164.9 KB)
Extracting: E:\boot\limine\limine-bios-cd.bin (24 KB)
Extracting: E:\boot\limine\limine-bios.sys (211.6 KB)
Extracting: E:\boot\limine\limine-uefi-cd.bin (2.8 MB)
Extracting: E:\boot\limine\limine.conf (322 bytes)
Extracting: E:\boot.catalog (2 KB)
Extracting: E:\oro-kernel-x86_64 (6.4 MB)
Extracting: E:\oro-limine-x86-64 (6.9 MB)
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:558B)
Using 'autorun.inf' label for drive E: 'ISOIMAGE'
Found USB 1.1 device 'MBED microcontroller USB Device' (0483:374B)
Device eliminated because it is smaller than 8 MB
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7628, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {610255A2-56A6-4FEE-9D8E-C1255C75A0A7}
Max parts: 128, Start Offset: 17408, Usable = 62746753536 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {AB9279B0-3DD0-4E76-AE96-788B622757F0}
  Size: 58.4 GB (62745673728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

closed 2025-03-04 by pbatard

Qix-, 2025-03-04:
Yep, aware how OSS works, was just asking if there was a way to disable it.

Is there any side effect to simply returning `0` from https://github.com/pbatard/rufus/blob/bc75274a9a585073f100e711227592691611e14c/src/process.c#L919 in all cases, just for my own local use? From what I can tell it's only used for diagnostics and robustness.

pbatard, 2025-03-04:
> Is there any side effect to (...)

You're asking me to provide some guarantee that if you modify the application in a way that it wasn't designed, it will still work as **you** expect. I'm afraid I can't satisfy that kind of request ever.

Best approach to removing sections you don't want to run is usually to comment them out, rather than alter their return value. And that's about as much help as I am willing to provide you with regards to your **custom, very specific** workflow.

Qix-, 2025-03-04:
Not asking for guarantees, just asking about the nature of that function. But thank you anyway.