pbatard/rufus #2715: Reconnecting a USB drive can lead to accidentally formatting the wrong drive

Raven-Singularity, 2025-03-28:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Log stuff is not applicable to this report.

Issue description
-----------------
Hello, great app.  Thanks for creating it!

I just have one small recommendation, which can help people avoid data loss.

Today I was trying to write an image to my USB stick, selected the IMG and my USB drive, and clicked Start.  Then I found out my USB key still had its read-only switch turned on, so I unplugged it, flicked the read/write switch, and plugged it back in... and almost hit Start, before noticing at the last moment that it had switched USB drives on me.  Rufus had automatically changed to my 128GB Ventoy boot disk (with over 60 operating systems collected on it), and I nearly wiped it all out.

It would be great if Rufus would take note of a USB drive disappearing, and _not_ fallback to another random USB drive when this happens.  Aside from disconnecting the USB drive to flip a read-only switch, sometimes USB controllers are just garbage and will randomly disconnect then immediately reconnect.  In such a case, it's quite likely someone wouldn't notice the USB drive they had selected had been changed.

You say in the FAQ that people should pay attention to what USB disk they select to avoid data loss -- and I did just that!  But Rufus changed the target disk by itself, unannounced, without my knowledge.  I am highly experienced with computers, and have been using them avidly since the late 1980s, but I made an assumption that the USB drive I had selected would stay selected.  It almost cost me a custom Ventoy theme and settings, and many hours of redownloading OS images.

The only reason I caught the mistake is because I always pedantically _double check_ before clicking Start in any formatting/partitioning program (sometimes triple check!).  I had already checked it once, and was sure it was still correct.

**Possible Solution 1:**
The drive that disappeared could either be greyed out, or just turned to a blank line in the drop-down list.  Once the USB drive is reattached, make it switch to the same drive again.

**Possible Solution 2:**
As a programmer myself, I just thought of an easier method to deal with the issue, which would avoid the need to keep track of USB drives disconnecting and reconnecting.  Simply have a blank entry in the list of USB drives.  When the currently selected USB drive disconnects, switch to the blank entry.  Done.

This would force the user to select a USB drive again, before formatting can start, and would eliminate any potential accidents.  It would be good if the blank entry was also the default selection, so Rufus would always require a manual drive selection.

If it's not too much hassle, please consider this small change -- I'm sure I can't be the only person that was tricked by USB drive reconnections in Rufus!

Cheers.

JonnyTech, 2025-03-28:
> Log stuff is not applicable to this report.

For completeness and reference, can you post your log with the unplug/replug actions included, please?

pbatard, 2025-04-09:
Okay, having considered this, I will not change the behaviour of Rufus, on account that the issue is based on expectations that are problematic and that will actually clash with other more reasonable expectations.

First of all the expectation that Rufus should automatically reselect the same device is you happen to unplug and replug it **is** problematic, because then this will run afoul of people who see that the device selected by Rufus is not the one they want to format, unplug it so that the device changes, and then replug their device after the change. If Rufus were to re-select the device then, I'm pretty sure I'd get the same kind of "accidentally formatting the wrong drive" but this time berating Rufus for re-selecting a device after it was unplugged. So I think we can safely dismiss the idea of automatic device reselection after re-plug as a bad idea.

Second, the suggestion of replacing the removed device by a blank or greyed out line is also problematic, because it goes against common expectations (used by every other modern computing application) that if you remove an element from a list, then that list gets reduced in real-time, to remove the element that is no longer available and reflect the change. Leaving blank/greyed entries in a list is problematic because it is unexpected and you are going to leave your users with some doubt as to whether the application is working properly (i.e. "Why doesn't the application just remove my device? Is there a possibility it will not be selecting the right device as a result?")

So I believe that the two reasonable expectations that anyone should have are:
- If one removes a device, then one should *reasonably* expect that the list of devices will be updated to reflect that change and also remove that device from the list, with the *logical* expectation that, if the device that was removed was the selected device, then the device selection **will** change.
- If one inserts a device, then one should *reasonably* expect that the list of devices will be updated to reflect that change and also add that device from the list, with the *logical* expectation that the currently selected device will not change, even if it so happens that the newly inserted device is one that was previously selected and removed.
- If devices are added/removed, the list should comprise only the devices currently present, with no blanks or extra addons (such as greying-out a removed device for instance, instead of removing it from the list altogether).

IMO, deviating from any of the expectations above is going to lead to problems with people who, I posit, **will reasonably** expect them to be be true, which is why I will **not** be doing any of the things you suggest, because while it *may* solve a potential issue with some type of interface users, it would most definitely create issues with another type of interface users, with the latter being the most likely types of users of Rufus (on account that they will expect Rufus to behave in a similar way as they see other Windows applications behave).

In short, the minute you unplug a device, **expect** change. If you then replug the device and expect not to have to validate that the device selected by Rufus is the one you actually want, then the issue is on you.

closed 2025-04-09 by pbatard