pbatard/rufus #2718: [FEATURE REQUEST] An option to enable or disable creation of the Windows RE partition

MechDR, 2025-04-03:
I am aware that this might be out of the scope of the project, but I've seen some things added to this project that I never thought I would see being added in any burn/mod tool, so I thought I'd ask here.

I usually make all my installs MBR (legacy boot), unless the hardware doesn't support that, so, naturally, I usually use the Windows IoT LTSD editions. Up until Windows LTSC 2021 you could prepartition the drive (single partition), make the partition active and just have point the installer to install there. In that case, the recovery "partition" was added on C:\ (which is what I preferred anyway) with just a simple `C:\Recovery` path.

That is not the case in Windows 11 LTSC 2024. You point the installer to install to a prepartitioned space and it shrinks the main partition to create the 625MB recovery partition.

In general, I wouldn't say that this is such a big deal. But, I work in IT and imaging tools are a huge part of my job. Some installs have to be precisely configured and maintained in that condition. We also use tools like DeepFreeze, Windows Update Blocker, etc., but, to be honest, there is less and less point in using tools like that in this day and age when things are changing by the minute, so you also need to unfreeze partitions every once in a while (a month or two) and let the software update itself. I did try to automate this, but the environment is so diverse, it got very complicated very fast. So, I opted for a simpler approach. Let the chips fall where they may with leaving most things update themselves and have a working image backup in case things go south.

Now, the problem is, since the environment is so diverse (old an new rigs, MBR, UEFI, pure UEFI, secure boot enabled/disabled, legacy PXE, UEFI PXE, etc.), I have to also use old versions of imaging tools that can boot in both MBR and UEFI (as well as PXE and UEFI PXE). Now, the kicker is, these old versions have problems with these partitions. Not that they don't recognize them, they usually recognize them just fine, but bringing them back to a blank drive can be an issue.

Now let's say the main drive died and I need to bring back an image to an empty media. Let's say the original drive was 250GB and the new one is 500GB. If not prepartitioned, imaging tools will usually just bring back sector by sector leaving 250GB unused. And this is not a real issue, you could either prepartition and just point to that partition and it will resize as well as update the MBR magic to where it needs to point to. You could also bring back the partition as is, MBR, one partition, active, and resize afterwards (even older tools know how to update the Windows MBR magic). But they have issues with the WinRE partition. It's not a standard NTFS partition, thus, some of them don't know what to do with it. For the time being, I've basically opted to use newer versions of partitioning tools (like PartitionGuru) that are DOS/FreeDOS based and thus, can be booted in both MBR and UEFI environments (with the right chaining tools of course), so that they can create the empty recovery parition, the main bootable one and I can just go "you go here, and you go there". But I feel like this is a complicated patch for something much simpler - just don't create the RE partition, create the RE environment on C:\, end of story.

Now I haven't done any real research on the issue, just a few basic Google searches, which were somewhat fruitful (found a reg key that seems to control not only if the RE is enabled or dsiabled, but also some other parameters), but I wanted to know if there is a simple fix for this (a few reg fixes, commands, etc.) if you're willing to take a stab at it. I could make a PR myself, but I'll need some guidance, I'm not good coder by any means.

pbatard, 2025-04-07:
Yeah, sorry, but I'm not going to look into that as it falls into the "custom specific workflow" category, and I am not designing Rufus to address custom workflows that very few people are expected to have.

See also https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want.

closed 2025-04-07 by pbatard

MechDR, 2025-04-08:
That's fine, I completely understand, it was a long shot anyway 😊.

Would you accept a PR related to this and maybe guide me on how to correct my code so that it can be merged?

MechDR, 2025-04-08:
OK, no problem 😊.

Thanks for this wonderful app BTW, it's invaluable to a huge number of people, including myself.