pbatard/rufus #2723: Win11 LTSC 26100.1742: Bypass online account does not work

Commifreak, 2025-04-11:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Using "Remove requirement for online account" with `de-de_windows_11_enterprise_ltsc_2024_x64_dvd_4f136f69.iso` does NOT work. There is no `Continue with limited setup` (Build `26100.1742`) visible during setup. Rufus is offering the option.

Log
---

[rufus.log](https://github.com/user-attachments/files/19702369/rufus.log)

Commifreak, 2025-04-11:
Iam disconnected. I also tried disabling my wifi. Even with this its not working. Microsoft wants me to install a driver for any network device.

No Ethernet built-in. Wifi-only

Commifreak, 2025-04-11:
> If that option isn't there, then microsoft might have changed something in the LTSC release that disables the bypass that Rufus uses

Exactly this. There is no option. Even no "I dont have internet".

I assume this build requires this new `ms-cxh://LOCALONLY` option.

pbatard, 2025-04-11:
Tested with `en-gb_windows_11_enterprise_ltsc_2024_x64_dvd_e2137661.iso` (which is the exact English International equivalent of `de-de_windows_11_enterprise_ltsc_2024_x64_dvd_4f136f69.iso`), made sure that I had no network connectivity, and I was properly prompted through the `I don't have internet' link to create a local account.

The issue is on your side, and not with Rufus. You need to make 100% sure that your network is disconnected.

Commifreak, 2025-04-11:
> The issue is on your side, and not with Rufus. You need to make 100% sure that your network is disconnected.

This i a blank-hard-drive notebook with only wifi. And I did not connected wifi. No „I dont have internet“.

I also disabled the wifi interface via BIOS. Now windows shows only the „Install driver“ button but still no „I dont have internet“.

🤔

pbatard, 2025-04-11:
Well, I tested that the bypass online account does work for Win11 LTSC 26100.1742 on a real system. And I saw the _I don't have internet_ option that allowed me to create a local account.

As much as you would like the issue to be with Rufus, __verifiable facts__ do say otherwise, since anybody can run the same test I did, and they will be able to see that the _I don't have internet_ option is available when the installation media is created with Rufus.

Therefore I can only reiterate that the issue is with your environment and that the remove requirement for an online account option of Rufus __does__ work and has not been broken by Microsoft with the latest Windows 11 LTSC images. I will not spend any more time on this issue.

Ibuprophen, 2025-04-11:
I had recently saw a tutorial regarding a clean install of Windows 11 (bypassing the Windows Login) on the following link. 

https://www.elevenforum.com/t/clean-install-windows-11.99/#step16

I hope it's okay with you @pbatard to share that link here 👍 

~Ibuprophen

pbatard, 2025-04-11:
@Ibuprophen, that's fine. Anything that can help is welcome.