pbatard/rufus #2724: Write error at sector 0: [0x00000005] on v4.7 while 4.2 is ok

N0rd1n, 2025-04-11:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Got error "Write error at sector 0: [0x00000005] Access is denied" on Rufus v.7, while no error on Rufus v4.2. And i don't have Controlled Folder Access or any incompatible software. 

Log
---
```
Rufus x64 v4.7.2231 (Portable)
Windows version: Windows 11 Pro x64 (Build 26100.3775)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'Samsung Type-C USB Device' (090C:1000)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 4
Disk GUID: {FAC57AD3-8C68-46AB-93CD-B897031E3BCC}
Max parts: 176, Start Offset: 32768, Usable = 32080144384 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Gap0'
  Detected File System: ISO9660
  ID: {FAC57AD3-8C68-46AB-93CC-B897031E3BCC}
  Size: 270 KB (276480 bytes)
  Start Sector: 64, Attributes: 0x1000000000000001
Partition 2:
  Type: EFI System Partition
  Name: 'EFI boot partition'
  Detected File System: FAT12
  ID: {FAC57AD3-8C68-46AB-93CF-B897031E3BCC}
  Size: 8 MB (8388608 bytes)
  Start Sector: 604, Attributes: 0x1000000000000001
Partition 3:
  Type: Apple HFS+ Partition
  Name: 'HFSPLUS'
  Detected File System: HFS/HFS+
  ID: {FAC57AD3-8C68-46AB-93CE-B897031E3BCC}
  Size: 1.5 GB (1562865664 bytes)
  Start Sector: 16988, Attributes: 0x1000000000000001
Partition 4:
  Type: Microsoft Basic Data Partition
  Name: 'Gap1'
  Detected File System: (Unrecognized)
  ID: {FAC57AD3-8C68-46AB-93C9-B897031E3BCC}
  Size: 300 KB (307200 bytes)
  Start Sector: 3069460, Attributes: 0x1000000000000001
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected EFI bootloader(s) (from '/efi.img'):
  ● 'bootx64.efi'
  Reported Grub version: 2.06-13+pmx6
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'PVE'
  Size: 1.5 GB (Projected)
  Uses: GRUB2 (2.06-13-pmx6)
  Uses: EFI
  Note: This ISO uses symbolic links, which may not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Note: DD image mode enforced since this ISOHybrid is not ISO mode compatible.
Using image: proxmox-ve_8.4-1.iso (1.5 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)

Write error at sector 0: [0x00000005] Access is denied.
Retrying in 5 seconds...

Write error at sector 0: [0x00000005] Access is denied.
Retrying in 5 seconds...

Write error at sector 0: [0x00000005] Access is denied.
Retrying in 5 seconds...

Write error at sector 0: [0x00000005] Access is denied.
Opened \\.\PhysicalDrive2 for shared write access

Found USB 3.0 device 'Samsung Type-C USB Device' (090C:1000)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {FAC57AD3-8C68-46AB-93CD-B897031E3BCC}
Max parts: 128, Start Offset: 17408, Usable = 32080165888 bytes

WARNING: 'Controlled Folder Access' appears to be enabled on this system
You may need to disable this feature, or add an exception, for Rufus to to work...

```

N0rd1n, 2025-04-11:
I successfully wrote image with v4.2

pbatard, 2025-04-11:
Most likely you have security software that whitelisted Rufus 4.2 but not Rufus 4.7.

There has been no functional change in the way Rufus writes to disk, and you are the only person reporting this (whereas Rufus is downloaded about to 3 million times each month, so if there was an issue like this due to Rufus, **a lot** of people would be reporting it instead of one person only).

What happens when you try from a different computer?

N0rd1n, 2025-04-11:
I've tried writing same img on other clean machine and everything was fine. 
So it should be some incompatable software indeed. Thanks!

Rufus x64 v4.7.2231
Windows version: Windows 11 Pro x64 (Build 26100.3775)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Samsung Type-C USB Device' (090C:1000)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {FAC57AD3-8C68-46AB-93CD-B897031E3BCC}
Max parts: 128, Start Offset: 17408, Usable = 32080165888 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Detected File System: NTFS
  ID: {BA2F9257-37DF-4DD7-8820-826125D763D6}
  Size: 29.9 GB (32079085568 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected EFI bootloader(s) (from '/efi.img'):
  ● 'bootx64.efi'
  Reported Grub version: 2.06-13+pmx6
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'PVE'
  Size: 1.5 GB (Projected)
  Uses: GRUB2 (2.06-13-pmx6)
  Uses: EFI
  Note: This ISO uses symbolic links, which may not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Note: DD image mode enforced since this ISOHybrid is not ISO mode compatible.
Using image: proxmox-ve_8.4-1.iso (1.5 GB)

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Opened \\.\PhysicalDrive1 for shared write access

closed 2025-04-11 by N0rd1n