pbatard/rufus #2731: iSO image extraction failure occurs many times, although I have downloaded many different ISO from the official website.

rongxuancheng, 2025-04-27:
### iSO image extraction failure occurs many times, although I have downloaded many different ISO from the official website.

**I ran it many times in different computers and replaced many different versions of Kingston USB sticks, and I got errors.**
Here's a screenshot of me

[image]

[image]

[image]

I hope you can explain to me why.
**Is the Chinese mainland version of the Windows system missing some c++ running files？**

_I didn't learn English well, please forgive me if there are some rude expressions and The meaning is ambiguous._

pbatard, 2025-04-27:
There is no issue with the chinese version and especially there is no possibility that it is missing runtime files.

Even if you don't speak English, you could use an automated translation engine to find out that you were asked to provide a log, and that your issue would be closed if you didn't provide one. I can not help you without a log, so I will close this issue.

closed 2025-04-27 by pbatard

rongxuancheng, 2025-04-28:
I will try it again and provide a log.