pbatard/rufus #2734: Forced NTFS

RandomTurtle, 2025-05-05:
my USB drive is 128Gb so it wants to format as NTFS but my laptop needs fat32, I create a 29G. partition but Rufus doesn't recognize the partition, it only sees the entire drive. I want Rufus to limit itself to the 29G partition. I have no smaller USB drives. Rufus should have a feature to partition the USB smaller to use fat32 but I can't find it.

closed 2025-05-05 by pbatard

RandomTurtle, 2025-05-05:
But when I create the Windows 11 USB with Rufus and press F12, it doesn't show up the USB in the boot selector. I asked ChatGPT and it said the cause is that it is NTFS, it needs to be FAT32. It is for Lenovo T470.

RandomTurtle, 2025-05-05:
Why did you close the issue so suddenly, it is not even resolved ??

pbatard, 2025-05-05:
> I asked ChatGPT 

ChatGPT is wrong. ChatGPT is not yet smart enough to replace expert advice, which is what I am giving you here based on actual proper human interpretation of the actual UEFI specs, rather than the scrapping of internet data that all too often falls short of accuracy, since it is generated by users who may not grasp the fundamentals of what they are trying to talk about.

>  when I create the Windows 11 USB with Rufus and press F12, it doesn't show up the USB in the boot selector.

Then you have a different issue than FAT32 vs NTFS. You probably have a Legacy vs UEFI detection issue. Or you did not enable Third Party certificates for Secure Boot. At any rate, this is not a Rufus issue, as I can guarantee that, unless that laptop **FAILS** UEFI certification, there is not an x86 based laptop is existence that will only boot from FAT32 and not NTFS. And that is not a ChatGPT illusion, that is a fact that is based on the absence of counter examples that would have come my way a long time ago, if there ever existed such a laptop or PC. Especially, the Lenovo T470 has been quite popular, and I can guarantee that if people had had trouble installing Windows using the NTFS partition created by Rufus for it, I would have had plenty of reports, whereas I have not had any.

> Why did you close the issue so suddenly, it is not even resolved ??

Because your issue is unrelated to Rufus. You are wasting your time thinking that you cannot boot from NTFS on your laptop. Instead you need to figure out what, in your settings, if preventing your laptop, which is able to boot from the NTFS drive created by Rufus, from being able to boot your drive. And that is not an issue I can resolve for you.