pbatard/rufus #2738: [suggestion] can you add a formatting button?

universish, 2025-05-13:
i created a usb flash drive almalinux installer with belenaetcher. windows file explorer does not see the usb flash drive.
flash drive partition 1 is EFI, partition 2 is full, with installation files.
windows 'create or format hard disk partitions' tool sees partition 1 but cannot delete or format it. partition 2 sees it as free space. I formatted partition 2. but file explorer didn't see it. so i ran rufus. i set Format Options without selecting the iso file. it erased and formatted the usb flash drive.
It would be great if there was a formatting option under iso when you click select in Drive Options. it would be great if we could use rufus as a formatting tool and have an option to convert usb installation memory to file storage memory.
with rufus it is very easy and hassle free to create an installation usb. the partitions are not invisible and unrecognizable. so windows file explorer recognizes the usb drive and we can format it. this shows the quality of rufus. but the other software is poor quality, simple software. some os's also recommend this software. we have to create a usb installation usb drive with these software. so formatting the usb drive becomes very difficult. thanks to rufus we can fix it. thanks. but there is no option to format the installation usb drive in rufus. so it is difficult to set it.
It would be nice if we could just click on Format in Drive Options and the other settings would be turned off and the formatting configurations would be configured automatically and if we set the Formatting Options configurations manually it would delete the partitions and then format them.

pbatard, 2025-05-13:
> it would be great if we could use rufus as a formatting tool and have an option to convert usb installation memory to file storage memory.

It's already there: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive

> but there is no option to format the installation usb drive in rufus.

There is. You just need to select `Non bootable` under _Boot selection_.

> It is impossible to format a USB drive without erasing the data on it

No: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_does_Rufus_need_to_delete_all_partitionsvolumes_when_creating_a_drive

Next time, please look if your questions have already been answered in the FAQ, as you were told to do when you opened the issue.

closed 2025-05-13 by pbatard

universish, 2025-05-13:
> There is. You just need to select Non bootable under Boot selection

ok, thanks