pbatard/rufus #2739: Not possible to save USB stick to VHD/XVHD, after finishing file disappear.

rebelx9x, 2025-05-21:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to save bootable Win11 USB stick to VHD or XVHD.
Process is smooth till it ends, and then file disappears from the folder I was saving it to.
tested on several computers, used VHD and XVHD, result is the same. File is visible during it's creation and disappears right after process ends. No error is displayed, just file is gone.
Log
---
```
Rufus x64 v4.7.2231 (Portable)
Windows version: Windows 11 Enterprise x64 (Build 22631.5335)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {5EF0FC5D-D1ED-4B85-B65E-A8C6DD0E4D9D}
Max parts: 128, Start Offset: 17408, Usable = 15479562752 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {62C5224C-A4D8-4792-BAB1-99483D26283E}
  Size: 14.4 GB (15477428224 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {A1963900-D702-444B-9756-81F55184B56A}
  Size: 1 MB (1064448 bytes)
  Start Sector: 30231432, Attributes: 0x9000000000000000
Found 5 additional UEFI revocation filters from remote SBAT
0 devices found
0 devices found
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB17EF11C
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.3 GB (7806648320 bytes)
  Start Sector: 2048, Boot: Yes

Save to VHD operation started
Saved 'C:\utilz\Disk_imaging\Windows\CGI-W11_x64_US-USB_bootable.vhdx'

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB17EF11C
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.3 GB (7806648320 bytes)
  Start Sector: 2048, Boot: Yes
0 devices found
0 devices found
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {5EF0FC5D-D1ED-4B85-B65E-A8C6DD0E4D9D}
Max parts: 128, Start Offset: 17408, Usable = 15479562752 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {62C5224C-A4D8-4792-BAB1-99483D26283E}
  Size: 14.4 GB (15477428224 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {A1963900-D702-444B-9756-81F55184B56A}
  Size: 1 MB (1064448 bytes)
  Start Sector: 30231432, Attributes: 0x9000000000000000
0 devices found
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB17EF11C
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.3 GB (7806648320 bytes)
  Start Sector: 2048, Boot: Yes

Save to VHD operation started
Saved 'C:\utilz\Disk_imaging\Windows\CGI-W11_bootable-USB.vhd'
```

pbatard, 2025-05-21:
I'm afraid I cannot replicate this issue. 

Also the lines:
``` 
Save to VHD operation started
Saved 'C:\utilz\Disk_imaging\Windows\CGI-W11_bootable-USB.vhd'
```

indicate that, as far as Rufus is concerned, the drive was successfully saved to VHD.

The process of saving to VHD/VHDX is also exceedingly simple: We just invoke a **single** Windows API called [`CreateVirtualDisk()`](https://learn.microsoft.com/en-us/windows/win32/api/virtdisk/nf-virtdisk-createvirtualdisk) and wait until Windows reports that it's either completed successfully or that there was an error. That's it.

Which means that Rufus doesn't even take care of actually writing the data to the file, and it especially has no say with what happens once the operation is completed. It's 100% Windows internal workings there.

Therefore, if the file disappeared, it's either a Windows internal bug (because I can guarantee that if you were to create your own application that invokes `CreateVirtualDisk()` you would see the same issue, since, again, that's all Rufus actually does here) or, much more likely, because you have a third party application that made that file disappear.

Be mindful that, depending of what you have on a drive, and since saving to VHDX is equivalent to scanning the full drive, an overzealous security solution such as an antivirus, may decide that the file contains a virus, and remove it altogether.

One test I would do to validate that is zero a USB drive (using <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus) and then, once this is completed save that drive to VHD/VHDX. Because the data on this drive will be all zeroes, there shouldn't be anything to 'trip' a security solution so this should validate whether it's the issue or not. Be minfdul that simply formatting or writing different content to a USB drive might not be enough, as (that is unless you uncheck "Quick format") there may still be data from previous writes, in unused sectors, that the security solution can pick up, so you must make sure the whole drive is overwritten with empty data for such a test to be conclusive.

At any rate, I really don't see anything I can do in Rufus here. If there was an error creating the file, Rufus would have reported it, and there is certainly **no** code in Rufus (yes, I double checked) that deletes the saved image. We don't even delete that image on user cancellation, but instead rely on `CreateVirtualDisk()` to do that for us.

In short, from what I can gather, your image was removed either because `CreateVirtualDisk()` has a bug, or because you have a security solution that thinks it contains malware.

pbatard, 2025-05-26:
No update from OP &rarr; Closed.

closed 2025-05-26 by pbatard