pbatard/rufus #2740: Crash with MSVC x64 Release build on Windows 11

ozone10, 2025-05-22:
Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

When building rufus Release x64 with MSVC resulting binary will immediately crash on launching, no window will appear after 
[image],
if running without config, with config it will silently crash.
Older version builds (commit with tags v4.7 and v4.6) also crash.
There is no issue on Windows 10 or x86, debug, MinGW, builds or action build https://github.com/pbatard/rufus/actions/runs/15118231274/job/42494216621

ref https://github.com/pbatard/rufus/pull/2713#issuecomment-2902231755

Log
---
```
00000075	0.00000000	[11796] *** Rufus init *** 	
00000076	0.00506230	[11796] Cur dir: 'D:\\' 	
00000077	0.00555010	[11796] App dir: 'D:\' 	
00000078	0.00579550	[11796] Sys dir: 'C:\WINDOWS\system32' 	
00000079	0.00581160	[11796] Usr dir: 'C:\Users\-----' 	
00000080	0.00593700	[11796] Dat dir: 'C:\Users\-----\AppData\Local' 	
00000081	0.00601040	[11796] Tmp dir: 'D:\Temp\' 	
00000082	0.00830480	[11796] PKI: Failed to get signature for 'D:\rufusp.exe': Cannot find the requested object. 	
00000083	0.00833540	[11796] Will use settings from INI file 	
00000084	0.00841640	[11796] found locale 'en-US' 	
00000085	0.06602310	[11796] loc file not found in current directory - embedded one will be used 	
00000086	0.06694180	[11796] localization: extracted data to 'D:\Temp\Ruf910D.tmp' 	
00000087	0.09011510	[11796] localization: found locale 'en-US' 	
00000088	0.09078490	[11796] localization: found locale 'ar-SA' 	
00000089	0.09124630	[11796] localization: found locale 'bg-BG' 	
00000090	0.09127510	[11796] embedded.loc(1040): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000091	0.09127510	[11796] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.	
00000092	0.09127510	[11796] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000093	0.09174760	[11796] localization: found locale 'zh-CN' 	
00000094	0.09212000	[11796] localization: found locale 'zh-TW' 	
00000095	0.09247940	[11796] localization: found locale 'hr-HR' 	
00000096	0.09278010	[11796] localization: found locale 'cs-CZ' 	
00000097	0.09317410	[11796] localization: found locale 'da-DK' 	
00000098	0.09344850	[11796] localization: found locale 'nl-NL' 	
00000099	0.09380290	[11796] localization: found locale 'fi-FI' 	
00000100	0.09416040	[11796] localization: found locale 'fr-FR' 	
00000101	0.09530340	[11796] localization: found locale 'de-DE' 	
00000102	0.09531920	[11796] localization: found locale 'el-GR' 	
00000103	0.09568720	[11796] localization: found locale 'he-IL' 	
00000104	0.09592990	[11796] localization: found locale 'hu-HU' 	
00000105	0.09619970	[11796] localization: found locale 'id-ID' 	
00000106	0.09621690	[11796] embedded.loc(6250): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000107	0.09621690	[11796] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.	
00000108	0.09621690	[11796] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000109	0.09645000	[11796] localization: found locale 'it-IT' 	
00000110	0.09669540	[11796] localization: found locale 'ja-JP' 	
00000111	0.09696040	[11796] localization: found locale 'ko-KR' 	
00000112	0.09724410	[11796] localization: found locale 'lv-LV' 	
00000113	0.09749510	[11796] localization: found locale 'lt-LT' 	
00000114	0.09751140	[11796] embedded.loc(8235): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000115	0.09751140	[11796] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.	
00000116	0.09751140	[11796] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000117	0.09774810	[11796] localization: found locale 'ms-MY' 	
00000118	0.09778580	[11796] embedded.loc(8625): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000119	0.09778580	[11796] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.	
00000120	0.09778580	[11796] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000121	0.09802160	[11796] localization: found locale 'nb-NO' 	
00000122	0.09826460	[11796] localization: found locale 'fa-IR' 	
00000123	0.09855260	[11796] localization: found locale 'pl-PL' 	
00000124	0.09879980	[11796] localization: found locale 'pt-BR' 	
00000125	0.09904820	[11796] localization: found locale 'pt-PT' 	
00000126	0.09929640	[11796] localization: found locale 'ro-RO' 	
00000127	0.09954950	[11796] localization: found locale 'ru-RU' 	
00000128	0.09982610	[11796] localization: found locale 'sr-RS' 	
00000129	0.10007420	[11796] localization: found locale 'sk-SK' 	
00000130	0.10032280	[11796] localization: found locale 'sl-SI' 	
00000131	0.10057070	[11796] localization: found locale 'es-ES' 	
00000132	0.10081760	[11796] localization: found locale 'sv-SE' 	
00000133	0.10105710	[11796] localization: found locale 'th-TH' 	
00000134	0.10139530	[11796] localization: found locale 'tr-TR' 	
00000135	0.10160950	[11796] localization: found locale 'uk-UA' 	
00000136	0.10189400	[11796] localization: found locale 'vi-VN' 	
00000137	0.10336080	[11796] Note: UAC is active 	
00000138	0.14477891	[11796] localization: initializing default message table 	
00000139	0.33744600	[11796] Rufus x64 v4.8.2248 (Portable) 	
```

ozone10, 2025-05-22:
removing `(uint8_t)strtoul(&p[1], &p, 10)` will stop rufus from crashing
https://github.com/pbatard/rufus/blob/be5642ef335379904bedf8b876ff6c291500f267/src/syslinux.c#L421

pbatard, 2025-05-22:
> removing `(uint8_t)strtoul(&p[1], &p, 10)` will stop rufus from crashing

Yes I understand that. But I either want to experience the crash myself, or have other people report that they see it, because, for all I can tell, this is seems to be tied to your specific environment, and again, I have not seen it once on the 3 different Windows 11 machines I tested.

I will be looking at it further, but if you are the only one experiencing it, this will be low priority.

Is it possible that you enabled additional Windows options or features on your machine, that could be the reason why (so far) you are the only one reporting this?

ozone10, 2025-05-23:
It is caused by [/Og](https://learn.microsoft.com/en-us/cpp/build/reference/og-global-optimizations?view=msvc-170) flag enabled by `/O2`.

Windows 11 24H2 has some new "improvements" in memory management and I recently updated to Visual Studio 17.14.

pbatard, 2025-05-23:
> and I've recently updated to Visual Studio 17.14

Well, I'm using 17.8.3, which is the latest up to date version of VS2022, and so would be GitHub, so, unless your report about also seeing the GitHub builds failing, inluding the 4.7 release one, was incorrect, I don't see how the version of Visual Studio being used would come into play. Also, if your VS installation reports 17.14, are you using a prelease of Visual Studio.

> Windows 11 24H2 has some new "improvements" in memory management

Do you have a source for that?

Before I treat this as a global issue (and I doubt Rufus will be the only application affected if that is really the case), I would really like to see other reports, as well as be able to replicate it, which I should, since I am fully up to date everywhere. It makes no sense for the issue you report not to be global if it's linked to a Windows update. Which means that we don't quite yet understand the issue. Which means we want to be cautious in rushing a "fix".

pbatard, 2025-05-23:
I have now broken down the line in 2 statements, as well as hardened our Syslinux version parsing while I was at it (though again, the previous code should have been fine). Hopefully this is the only place in the code where the new MSVC way of optimizing things will play tricks on us...

Thanks again for reporting this issue (and making me realize my VS 2022 installation was quite outdated).

closed 2025-05-23 by pbatard

ozone10, 2025-05-23:
Version 17.14.0 is quite new (only 10 days old) and github action still use version 17.13.

It is interesting that there is no issue on Windows 10.

For memory management and other security "improvements" I cannot find source I read, and MS released only some marketing text that 24H2 has better performance. But there are some interesting articles about MS changing some internal functions in 24H2.

https://www.linkedin.com/pulse/why-windows-11-24h2-goes-bsod-vb-audio-software-x8use
https://cookieplmonster.github.io/2025/04/23/gta-san-andreas-win11-24h2-bug/

commit 207a3309 (2025-05-23): [core] fix a crash due to overzealous MSVC optimizations

* Visual 2022 version 17.14.2 changes the way it performs optimizations with /O2
  which results in innocuous C statements, such as invoking strtoul() twice on
  the same line, and using the pointer returned by the first call in the second,
  crashing the application with a read access violation (when built as RELEASE).
* To alleviate that, we break down the Syslinux strtoul() invocation, as well as
  harden our Syslinux version processing while we're at it.
* Hopefully this is the only line of code where the new mode of MSVC optimisati [... cut]