pbatard/rufus #2751: build failure on MSYS2 since v4.7

ognevny, 2025-06-05:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

starting with version 4.7, Rufus fails to compile on MSYS2 with our [recipe](https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-rufus/PKGBUILD)(the only thing changed is a version number) fails with such errors:

```
     CCLD     rufus.exe
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lcrypt32-delaylib: No such file or directory
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -ldwmapi-delaylib: No such file or directory
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lsetupapi-delaylib: No such file or directory
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lversion-delaylib: No such file or directory
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lvirtdisk-delaylib: No such file or directory
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lwininet-delaylib: No such file or directory
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lwintrust-delaylib: No such file or directory
  collect2.exe: error: ld returned 1 exit status
```

it seems that `-l<lib>` is passed literally as the name to `-l`

Log
---
\-

pbatard, 2025-06-05:
Looks like an ucrt64 issue. I'm afraid I don't support building for ucrt64 yet (and I am not sure if I am ever going to bother to as long as mingw64 works fine).

I **might** accept a patch that makes our current build scripts compatible with ucrt64, as long as they aren't too intrusive or look maintenance heavy.

ognevny, 2025-06-05:
see a [full](https://github.com/msys2/MINGW-packages/actions/runs/15470029133/job/43551692775?pr=23923) GHA job

ognevny, 2025-06-05:
the issue is encountered in [this PR](https://github.com/msys2/MINGW-packages/pull/23923). GHA job is run on clean msys2 environment, recipe link is put above. it shouldn't differ from Rufus CI actually...

pbatard, 2025-06-05:
The other thing I see is that you seem to be invoking the build from outside the project root directory so you get some directory prefixing:
```
  make[1]: Entering directory '/d/_/B/src/build-MINGW64/src'
  Making all in ../.mingw
  make[2]: Entering directory '/d/_/B/src/build-MINGW64/.mingw'
    SED      ../../rufus-4.8_BETA/.mingw/crypt32.def
    LIB      crypt32-delaylib.lib
    SED      ../../rufus-4.8_BETA/.mingw/dwmapi.def
    LIB      dwmapi-delaylib.lib
    SED      ../../rufus-4.8_BETA/.mingw/setupapi.def
    LIB      setupapi-delaylib.lib
    SED      ../../rufus-4.8_BETA/.mingw/version.def
    LIB      version-delaylib.lib
    SED      ../../rufus-4.8_BETA/.mingw/virtdisk.def
    LIB      virtdisk-delaylib.lib
    SED      ../../rufus-4.8_BETA/.mingw/wininet.def
    LIB      wininet-delaylib.lib
    SED      ../../rufus-4.8_BETA/.mingw/wintrust.def
    LIB      wintrust-delaylib.lib
```

Whereas when built from the current project directory, one gets:
```
Making all in ../.mingw
make[2]: Entering directory '/d/a/rufus/rufus/.mingw'
  SED      crypt32.def
  SED      dwmapi.def
  SED      setupapi.def
  SED      version.def
  SED      virtdisk.def
  SED      wininet.def
  SED      wintrust.def
  LIB      crypt32-delaylib.lib
  LIB      dwmapi-delaylib.lib
  LIB      setupapi-delaylib.lib
  LIB      version-delaylib.lib
  LIB      virtdisk-delaylib.lib
  LIB      wininet-delaylib.lib
  LIB      wintrust-delaylib.lib
```

My guess is that the `####-delaylib.lib` files are written to your current build dir instead of the `.mingw/` dir. Or that they are placed in a location where you can't find them at linktime.

I don't really remember changing anything that should affect non root builds, but I'll see if I can look at those. It'll probably be a few days before I do that though, so if you locate the issue before that and can send a patch, feel free to do so.

ognevny, 2025-06-05:
> The other thing I see is that you seem to be invoking the build from outside the project root directory so you get some directory prefixing

yes, this is a kind of requirement: it's better to have separate build dirs for each environment for better local testing

pbatard, 2025-06-05:
No worries, I don't have an issue with that, and I do want Rufus to support those builds as well. But I don't really test them on a regular basis, so breakage may happen, and fixing them won't be a high priority (though I do want to fix them... eventually).

commit fe9b1379 (2025-06-09): [wue] forward wrapper parameters to original setup.exe

* Also fix MinGW builds from outside of root directory
* Closes #2751
* Closes #2752
closed 2025-06-09 by pbatard