pbatard/rufus #2756: Recent GitHub update prevents all downloads from https://rufus.ie

pbatard, 2025-06-15:
Well, in an **INCREDIBLY STUPID** move, GitHub have recently decided to update their servers to **remove** the ability to obtain the Content-Length of a GitHub hosted file, when `Accept-Encoding: gzip, deflate` is being used.

This has the result of completely **BREAKING** Rufus' ability to download any files from https://rufus.ie (since, in an effort to save GitHub some bandwidth, we allow compression for downloads by default, so yeah, thanks for not playing ball in return Github!), and worse, it prevents every existing version of Rufus (outside of the Windows Store) to receive program updates... ever.

We're going to try and see if we can get GitHub to revert that change (not at all optimistic about it), but in the meantime, we're going to have to rush a 4.9 BugFix release, while knowing that people won't be able to automatically update Rufus to that version... 😦

_"YOU GITHUB MANIACS, YOU BLEW IT UP! DAMN YOU ALL! DAMN YOU TO HELL!!!"_

pbatard, 2025-06-15:
NB: This might have more to do with Microsoft than GitHub (though with GitHub being owned by Microsoft, it's pretty much the same difference), since `curl -H 'Accept-encoding: gzip, deflate' -L -I https://rufus.ie/files/syslinux-6.04/ldlinux.c32` does appear to return the `content-length` data, yet `HttpQueryInfoA(..., HTTP_QUERY_CONTENT_LENGTH, ...)` with the same `Accept-encoding` does not work and returns error `Unable to retrieve file length: [0x00002F76] The requested header was not found`.

closed 2025-06-15 by pbatard

pbatard, 2025-06-16:
Ah, this is a different issue, which I have now fixed. Thanks for reporting it. Rufus does **not** use the links from https://rufus.ie/downloads/ for its updates. Instead, this page is edited manually to provide an alternate download page, but this is not the actual location where the executables are hosted, so typos in it don't matter outside of people visiting it for an alternative source for downloads (though of course, it's a lot better when links from this page work too!).

pv7721, 2025-06-16:
Oh, I get it now: you meant updates from WITHIN rufus itself were broke by GH? As I seldom use it, I always preferred to download the latest installation kit (from the /downloads page) . I'm happy that I could spot the typo. Thank you for this great software!

commit 3791b96f (2025-06-15): [net] fix downloads no longer working after the recent GitHub server update

* GitHub appears to have updated their HTTP servers to enable gzip compression when requested.
* Unfortunately, the default from Rufus was to attempt to use compression where possible
  (ironically in an effort to save GitHub some bandwidth), which doesn't seem to sit too well
  with Windows' HttpQueryInfoA(HTTP_QUERY_CONTENT_LENGTH) API, as it errors out on trying to
  produce the content-length *DESPITE* the field being reported by GitHub in the HTTP header.
* As a result, we switch all HTTP downloads *not* to try t [... cut]