pbatard/rufus #2759: Rufus 4.9.2256

carey934, 2025-06-19:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
How can I use Rufus to create a bootable MicroSD card for my NanoKVM Lite? It appears I must download a .xz file that must be uncompressed to a .img file. (https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0)  But when attempting to write that file via DD on Rufus, no DD option ever appears and after the normal write is complete, my micro SD card is not bootable.

However, if I use a Rufus alternative, balena Etcher, it accepts the .xz file and is intelligent enough to decompress it and write it as a DD image automatically. Then the micro SD works perfectly and boots in the NanoKVM.

Is there any way for Rufus to mimic the functionality of balena Etcher for ease of use and compatibility? Or has Rufus stopped further development towards compatibility and ready to be replaced with a more modern, free alternative that seems to work better? (I have been using Rufus for a very long time as my go-to tool, but now it is letting me down. When I come for answers the developers are very defensive and do not like criticism. What am I doing wrong or is it time for me to move on?)

Log
---
```
<FULL LOG>
```

pbatard, 2025-06-19:
No log. You were warned.

I see **no** issue opening and writing the `20250217_NanoKVM_Rev1_4_0.img.xz` image with Rufus. There is not need to uncompress it.

Therefore I'm going to assume that the fact that you did not provide a log means that you did not really try to write the image with Rufus.

closed 2025-06-19 by pbatard

pbatard, 2025-06-19:
Fot reference (to demonstrate that Rufus does properly write the `20250217_NanoKVM_Rev1_4_0.img.xz` **unmodified** to the USB drive.

* Disable the automatic creation of `System Volume Information\` on removable drives by following the steps described [here](https://superuser.com/a/1199824/286681) and reboot (because otherwise, Windows automatically alters the drive data to create extra folders, but this has **nothing** to do with Rufus, and it has no impact on the drive boot behaviour anyway)
* Write from Rufus. Here is my full log:
```
Rufus x64 v4.9.2256
Windows version: Windows 11 Pro for Workstations x64 (Build 26100.4351)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it matches Disk GUID {B082BAF2-8033-47A5-BF72-965079A1DE96}
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
Using 'autorun.inf' label for drive G: 'boot'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0000DE79
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.9 GB (16012877824 bytes)
  Start Sector: 2048, Boot: Yes
Found 5 additional UEFI revocation filters from remote SBAT
Found 4 active Secure Boot certificate entries from remote
Found 1 revoked Secure Boot certificate entries from remote
Local time: 13394804009
Next update check in 84647 seconds.
Scanning image...
ISO analysis:
  'D:\Imgs\20250217_NanoKVM_Rev1_4_0.img.xz' doesn't look like an ISO image
Disk image analysis:

Error: unsupported check; not verifying file integrity
  Image is a compressed bootable disk image
Using image: 20250217_NanoKVM_Rev1_4_0.img.xz (189 MB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive8 for exclusive write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
Writing compressed image:

Error: unsupported check; not verifying file integrity
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Opened \\.\PhysicalDrive8 for shared write access
Remounted \\?\Volume{709ff46d-4cfb-11f0-9233-88d7f6e01c0c}\ as G:

Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it matches Disk GUID {B082BAF2-8033-47A5-BF72-965079A1DE96}
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xBAFB6BF3
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT16
  Size: 16 MB (16777216 bytes)
  Start Sector: 1, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 32769, Boot: No
```
The `Error: unsupported check; not verifying file integrity` is actually not an error but a mere warning that some features of XZ compression will not be supported. Unless the BusyBox decompression library starts to add support for that feature I'll probably silence that in a future version of Rufus since, as you can see below, it has no impact on the ability of Rufus to properly write XZ compressed images. 
* Compare the USB written data to the original file. I've done that from a Linux system since it's a bit easier, through the following commands:
```bash
# unxz 20250217_NanoKVM_Rev1_4_0.img.xz
# cmp -b /dev/sdc 20250217_NanoKVM_Rev1_4_0.img -i 512 -n $((`stat -c %s -- "20250217_NanoKVM_Rev1_4_0.img"` - 512))
#
```
The `stat -c %s -- "20250217_NanoKVM_Rev1_4_0.img"` part takes the size of the uncompressed image file. The `-i 512` and $(( ... - 512))` is just there to skip the MBR, as Windows stupidly adds a checksum there at offset `0x1B8` automatically, and I have not yet found how to disable this behaviour. Just like the `System Volume Information\` directory, this checksum has no impact on the drive's ability to boot, and you can easily compare that the other bytes of the MBR are the same as the ones from the image.

**Therefore** (outside of Windows' obnoxious automated behaviours, but you will get the same from balenaEtcher on Windows if you plug a USB drive), Rufus does write `20250217_NanoKVM_Rev1_4_0.img.xz` exactly as expected and there is nothing from your report about Rufus having to _"mimic the functionality of balena Etcher for ease of use and compatibility"_. As we have demonstrated above, Rufus is writing the data just as well as Etcher or any other `dd` like utility, and it has no trouble using the compressed `.xz` image as source.

carey934, 2025-06-20:
I realize you want logs and I respect that. I also appreciate your response. Perhaps I should have been more clear in that Rufus does write the image, but the disk will not boot. I have Googled for more information and consistently find a complaint that Rufus has a reputation of not creating Linux boot disks correctly in that they will not boot. The overall recommendation is to use a different tool for consistency, while Rufus may work sometimes, sometimes it does not.

Given I normally create Windows tools, I appreciate that Rufus auto detects a Windows installer ISO and offers me the options of altering that media to adjust installation defaults (such as requiring an online microsoft account) and I greatly appreciate the effort that went in to this design. Imagine my surprise when I learned the down arrow attached to the SELECT button, was in itself, a completely different button with online downloads of Windows ISOs behind it. This design choice is quite hidden and not intuitive since the down arrow and the select button are attached, without any separation, this does not follow standard design most software voluntarily complies with. I wish I had known just by looking at the screen, those were two separate functions. Just a suggestion.

As a professional computer technician, educator, author, radio show host and YouTube content creator with over 400,000 subscribers, I want you to know I do not mean to criticize or troll. Nor am I a simple enthusiast. I also have much to learn and don't claim to know everything. I assure you, I am here to get educated and offer some consideration to potentially improve Rufus.

It does bother me that the Linux community likes to use computer terminology incorrectly, referring to writing data as 'burning' or 'flashing' when the correct term is simply 'writing' data to a flash drive, or burning software to an optical drive, or flashing data to an eeprom.

But, as a corporate IT drone, perhaps my intolerance is not a fair judgement of the perception of intelligence from those who use the terms improperly.

Having said that, I cannot tell if Rufus is writing the data in ISO mode or in DD mode. I wish I knew. I wish I didn't have to hit the START button to see if there were more options, because then it's not really a START button any more, but rather a PROCEED button.

I don't understand the difference between writing an image file in DD mode vs ISO mode. I can only tell you from my experience, some images require DD mode and others require ISO mode and I only know that because I used to be able to tell Rufus (a long time ago) which way *I* wanted the file written.

I am sad to lose the control, not be given any information to the process that is happening and have the image not work afterwards with nothing more I can try since all of my options have been removed.

I understand that a HybridISO will offer the option of DD mode writing, but WHAT IF the HybridISO was not made properly? Its not Rufus's fault the creator of the image file is not following accepted standards. Heck, if they did that, they wouldn't call writing to flash memory 'flashing' or 'burning' but they do it anyway.

Rufus doesn't seem to allow any 'advanced' mode for those of us who would like to over-ride the defaults in such circumstances. But, because we can't, we are forced to use Rufus alternatives to get the job done. That makes me sad.

If you like, I can make a video demonstrating this as I try to make a bootable MicroSD card for my NanoKVM with Rufus and it refuses to boot. Then re-write the image on to the same MicroSD card with Etcher and it works perfectly.  Then I see people in the Linux forums this is typical behavior from Rufus and they no longer recommend Rufus.

I suppose this feels personal, because I love Rufus and never had an issue with it.... Until recently, I was not able to relate. But the issue isn't whether or not Rufus writes the image. Yes, Rufus does write the image, However, the image will not boot. I have only experienced this with Linux-based bootable images, but not all Linux-based bootable images. With Windows bootable images, Rufus always works 100%. 

Again, I appreciate your attention to this, and my intention is to better understand this weird discrepancy. 

Thank you!
Carey Holzman