pbatard/rufus #2760: ARM64 Windows 11 24H2 doesn't see any drivers even I already integrated them before.

xfacts78, 2025-06-20:
Hi. I created ARM64 Windows 11 24H2 iso from uupdump and tried to install on 128 GB USB 3.0 pen drive .

[https://uupdump.net/download.php?id=73b4b15d-fdfc-4a19-977d-248e76c55468&pack=en-us&edition=core](url)

I modified the ConvertConfig.ini to add all the necessary drivers to the .wim file. Later I decided to split to 4GB less for the FAT32 compatibility since I decided to replace in the Microsoft Surface Pro 11 ARM64 factory image . I did manage to install my custom windows 11 ARM 24H2 and even 25H2 Beta one on the Surface device. 

Also I managed to extract to MSI file format drivers and firmware to integrate to the iso and I did it too. 

I need to mention small but an important detail : I just copied the to sources folder expect boot.wim file which is original Microsoft Surface's file and I guess it's also a structure of MS bootable interface installation and it worked well. 

I used RUFUS 4.9 and all the MS Surface ARM drivers integrated properly. 

The problem is I can create the USB stick but I cannot use RUFUS even as FAT32 file system in my Surface ARM device which showed me this :

[image]

The log for creating as a NTFS based system and this manner I cannot go beyond above picture. I even cannot install via second usb port or even as a separated folder which all drivers I cannot see the manually install internal SSS storage etc.

I can confirm the integration of all drivers since I use it as a Microsoft recovery image replacing with different builds like custom iso as 24H2 or Beta of 25H2. 

[rufus.log](https://github.com/user-attachments/files/20839805/rufus.log)

Thanks in advance .

pbatard, 2025-06-20:
>  Later I decided to split to 4GB less for the FAT32 compatibility

Why? There's no need to do that and apart from some ill-fated ARM32 (not ARM64) devices 10 years ago, all current UEFI based devices are compatible with UEFI:NTFS (the method used by Rufus to boot from NTFS) and therefore there is zero need to want to use FAT32, even with Secure Boot enabled. You may have to enable 3rd party UEFI CA for Secure Boot (as should **always** be the case unless you want to be **restricted** by Microsoft with regards to usage of **your** device), but that's about it.

> which showed me this:

Well, you are lacking a disk driver for your device, and I'm afraid it is not Rufus' job to provide it.

**Generic** ARM64 Windows 11 ISOs are unlikely to contain all the disk drivers you need, so you will need to see with the manufacturer of the device to obtain that driver (yes, even if the manufacturer of the device is Microsoft).

I'm afraid that this is not a Rufus issue. I'll leave this issue opened for a while if people want to chime in, but since it is a driver issue related to your specific hardware, that **should** be solved by the manufacturer of your hardware, I'll probably close it in 24 hours.

xfacts78, 2025-06-20:
> The only advice i can give is to download an official ISO and inject the drivers using the command line (microsoft points you to instructions on how to do that on the download page, you do have to extract the files from the ISO first)
> 
> You would have to manually copy the files to a USB after injecting the drivers though

It seems you do not know about Microsoft's Surface recovery image structure. I have been using MS Surface PC for years and there isn't any official iso for any Surface PC product. Microsoft uses FAT32 system that's the reason I have to follow / stick with FAT32 file system and I did so accordingly too and success. Unofficially we can create ISO with all drivers and works pretty well and my initial aim was to make a ISO with drivers but I cannot manage to success. 

Actually I tried Ventoy but no success .

pbatard, 2025-06-20:
>  I want to use secure boot to be enabled always

And I did **not** ask you to disable Secure Boot in any way. I just asked you (if necessary) to enable Windows + Third Party CA Secure Boot, since Microsoft has recently created a two tier Secure Boot system, and, whereas the UEFI:NTFS solution Rufus uses **is** signed for Secure Boot, it uses the tier that Microsoft leaves for "peasants" (i.e. everybody who isn't Microsoft, even if they do qualify for UEFI Secure Boot certification) and that, once you realise that Microsoft were caught with their pants down on BlackLotus and had to revoke ALL of their Secure Boot signed Windows UEFI bootloaders (even as they claimed that the reason they created two-tier system was really because Windows bootloaders should be more trustworthy than non Windows ones), is really aimed at **restricting** what people can run on Microsoft-controlled devices.

> So RUFUS cannot create a working ARM64 usb bootable devices.

That's not what I have experienced and other people have reported.

Rufus is compatible with Secure Boor on ARM64 devices, and it can very much create ARM64 bootable media, for Secure Boot enabled ARM64 devices.

> So you are saying RUFUS does not support ARM64 especially for Surface devices?

I don't know how you interpreted what wrote, but this is the exact opposite of what I have actually been saying. Even when using UEFI:NTFS, Rufus can create Secure Boot compatible Windows installation media for ARM64 based devices. So if your device cannot install boot or install Windows, then it's either because it's not fully UEFI compliant or you are missing drivers.

xfacts78, 2025-06-21:
> And I did **not** ask you to disable Secure Boot in any way. I just asked you (if necessary) to enable Windows + Third Party CA Secure Boot, since Microsoft has recently created a two tier Secure [...]

I'll try again if I did miss some steps or not. Thank you for your effort here and your application.

xfacts78, 2025-06-23:
I realized that current implementations of making .iso images from uupdump.net has a issue with Surface ARM64 devices and probably other ARM64 Windows PCs too. I tried different manners but without success .

As a result I'd like to close my case. I'm a newbie to this platform . Thanks everyone of helping .

closed 2025-06-23 by xfacts78