pbatard/rufus #2764: deactivated checkboxes look active checkboxes in dark theme

CennoxX, 2025-07-02:
Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
There is a minor issue with the dark theme in Windows 10 @ozone10.
This is what deactivated checkboxes look like in light theme:

[image]

In the dark theme from the newest build of commit e571996525bc6ea5ad43b829df1c10fff6c89395 it is very difficult to see, if the checkbox is active or not:

[image]

As a reference this is what a deactivated checkbox looks like in Notepad++

[image]

Log
---
```
Rufus x64 v4.10.2264 (Alpha)
Windows version: Windows 10 Pro x64 (Build 19045.5965)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
0 devices found
Found 5 additional UEFI revocation filters from remote SBAT
Found 4 active Secure Boot certificate entries from remote
Found 1 revoked Secure Boot certificate entries from remote
```

ozone10, 2025-07-02:
@pbatard 
https://github.com/pbatard/rufus/blob/e571996525bc6ea5ad43b829df1c10fff6c89395/src/ui.h#L42

seems to use same colour as normal text, IIRC it should be `RGB(0x80, 0x80, 0x80)`

CennoxX, 2025-07-02:
> [@pbatard](https://github.com/pbatard)
> 
> [rufus/src/ui.h](https://github.com/pbatard/rufus/blob/e571996525bc6ea5ad43b829df1c10fff6c89395/src/ui.h#L42)
> 
> Line 42 in [e571996](/pbatard/rufus/commit/e571996525bc6ea5ad43b829df1c10fff6c89395)
>  #define DARKMODE_DISABLED_TEXT_COLOR				RGB(0xE0, 0xE0, 0xE0) 
> 
> seems to use same colour as normal text, IIRC it should be `RGB(0x80, 0x80, 0x80)`

You are right, changing to `RGB(0x80, 0x80, 0x80)` fixes it:

[image]

pbatard, 2025-07-02:
Yeah, I screwed up the value when I refactored this part of the PR to move all the colour defs back to ui.h, where we already had most of them. I'll fix this soon(ish).

commit c6a85b09 (2025-07-03): [ui] fix wrong colour for disabled text in dark mode

* Closes #2764.
* Also fix signature not being properly applied to MinGW GitHub Actions ALPHA builds.
closed 2025-07-03 by pbatard