pbatard/rufus #2770: Read/Write from network drive

RFBomb, 2025-07-16:
Checklist
---------
- [X] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [N/A] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [N/A] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [N/A] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [X] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Use case: 

create an image file of microSD card of unknown proprietary format (this works fine) and store it onto a network location (instead of C:). Then take that image file and write it to a new microSD card. 

The issue is that Rufus does not seem to support writing or reading to/from non-local drives. I even attempted to manually paste in the file path and it simply acted like I hadn't selected a file. 

Note: this is an industrial/corporate use case, and I'm exploring options for both command line (already read the FAQ) and ease of use for our technicians. 

I'm open to assisting with command line integration if needed (time permitting). We have a utility that mirrors locally already, so the network access portion wouldn't be a problem for techs if command line is available (we'd just wrap Rufus). But for the engineers not having read/write to the network would be cumbersome for testing and maintained of our images. 

To summarize: is there any technical reason Rufus doesn't support network drives? And is this something you'd look into/offer some assistance if I were to initiate a PR ?

Log
---
```
not included - no log produced 
```

RFBomb, 2025-08-16:
I would disagree that the issue is resolved. RUFUS can not save to any network drives that I or my coworkers use. HDD-Raw Copy (which also requires to run as ADMIN because it does the same task), can read and write to the network drives without issue. RUFUS simply acted as if an empty string was pasted in instead of a legit path. I also don't recall being able to browse network drives. 

My user account is an admin account on my computer, and these are mapped network drives that it could not see or interact with. 

If you do not wish to support, fine, its your development time. But the issue itself was not due to account privileges, when this was the only application unable to access those paths.

pbatard, 2025-08-16:
> RUFUS simply acted as if an empty string was pasted in instead of a legit path.

Please provide a full log then, as you were asked to do. I am not seeing the behaviour you describe, even when using the Microsoft Store version. As a matter of fact, I have no issue opening image from network shares in **my** environment.

What happens exactly when you attempt to use a network path? And what makes you say that Rufus acted as if an empty string was pasted?

Please don't complain about developers dismissing your issue when you **choose** not to provide the required amount of data to investigate (especially never assume that because you see a problem in your environment, then everybody else will see it).

pbatard, 2025-08-16:
I will also point out that you did check:

> - [X] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.

pbatard, 2025-08-29:
Again, I have no issue accessing content from network drive in my environment.

**Therefore** your issue is environmental.
**Therefore**, since it's virtually impossible for me to investigate it further **since I do not see your issue at all**, I am not planning to investigate it further.