pbatard/rufus #2781: Rufus uses the Internet without permission!

Andrei030x, 2025-07-28:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I have disabled the update check. However, when choosing the iso image is trying to send something to different ip addresses. (but of course the firewall blocks it)

In your FAQ and message in the logs:
> Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.

It is clearly stated that there should be no network activity when disabling updates. I'm waiting for an explanation.

Log
---
```
<FULL LOG>
Rufus x64 v4.9.2256
Windows version: Windows 10 Pro x64 (Build 19043.1586)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0419 (ru-RU)
Detected installed Windows Language Pack for 0x0409 (English (English))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected GRUB version: 2.12 (from '/boot/grub2/i386-pc/normal.mod')
  Detected Shim version: 15.8 (from '/EFI/BOOT/BOOTIA32.EFI')
  Detected Shim version: 15.8 (from '/EFI/BOOT/BOOTX64.EFI')
  Detected Shim version: 15.8 (from '/EFI/BOOT/bootx64.efi')
  Detected GRUB version: 2.12 (from '/EFI/BOOT/grubx64.efi')
  Supported GRUB filesystems: ext2, f2fs, fat, hfsplus, iso9660, gzio, xfs
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
UEFI bootloaders analysis:
  • /EFI/BOOT/BOOTIA32.EFI
  (Unsigned Bootloader)
  • /EFI/BOOT/BOOTX64.EFI*
  Signed by 'Microsoft Corporation UEFI CA 2011'
  Note: This bootloader may fail Secure Boot validation on *some* systems,
  unless you enable "Microsoft 3rd-party UEFI CA" in your 'BIOS'.
  • /EFI/BOOT/bootx64.efi*
  Signed by 'Microsoft Corporation UEFI CA 2011'
  Note: This bootloader may fail Secure Boot validation on *some* systems,
  unless you enable "Microsoft 3rd-party UEFI CA" in your 'BIOS'.
  • /EFI/BOOT/grubx64.efi
  Signed by 'Fedora Secure Boot Signer'
  • /EFI/BOOT/mmx64.efi
  Signed by 'grub2-signer'
ISO label: 'Fedora-WS-Live-42'
  Size: 2.2 GB (Projected)
  Uses: GRUB2 (2.12-nonstandard-gdie)
  Uses: EFI
Using image: Fedora-Workstation-Live-42-1.1.x86_64.iso (2.2 GB)
```

ozone10, 2025-07-28:
@pbatard 
initially I could not reproduce it, but after deleting registry with Alt+R and adding rule in firewall to block Rufus I could reproduce it.

After some quick git bisect, it seems to be commit https://github.com/pbatard/rufus/commit/688f011f317492225ded10b51edc19b39d62fbd4 where "Enable download of remote active/revoked Secure Boot certificate thumbprints" was added.

Andrei030x, 2025-07-29:
Well, in the meantime, I reproduced this on another computer, also Windows 10, simplewall firewall.

Why don't you try this with Fort Firewall or simplewall? Both are free and open source. The problem might be in determining which program the traffic is coming from.

pbatard, 2025-07-29:
Fort Firewall requires disabling Core Isolation, and I got an error about it when I tried to install it. I'm not willing to disable it, when other firewalls and network traffic loggers don't require it.

And I just tried simplewall, and saw no attempts at outbound connections when the check for updates is disabled (be very mindful that you **MUST** make sure that Check for updates is disabled in your Rufus settings, or else you **will** see network traffic).

And seriously, I have now tried r different network logging applications, where I double checked that, when the check for updates is enabled or ran manually, I got traffic logged as expected (which means that it is **not** a firewall misconfiguration or limitation on my side).

If you do see traffic, then I will ask you to please provide:
1. The full log from Rufus.
2. The IPs (or URLs) the application connected to.

Without this, I see simply too many chances that you somehow enabled the check for updates when you ran your tests and got traffic as a result.

pbatard, 2025-07-29:
Also, the documentation for [`CertGetCertificateChain()`](https://learn.microsoft.com/en-us/windows/win32/api/wincrypt/nf-wincrypt-certgetcertificatechain) definitely implies that there can be CRL/OSCP **online** revocation checks carried out. But, even if we wanted to, I am not seeing any flag we can pass to try to disable those. `CERT_CHAIN_REVOCATION_CHECK_CACHE_ONLY` looked like a good candidate, but it logically seems to require that something else populated the cache, and I therefore would expect errors if that is not the case.

Now, since I also expect that the OSCP/CRL checks being carried out won't be that relevant for Secure Boot (since it's a different trust chain than the usual Windows one, where revocations are not really handled by OSCP/CRL) I might see if there is an alternate way of getting to issuer of the certificate without invoking `CertGetCertificateChain()`. But that will be very low priority.

ozone10, 2025-07-29:
I think Vodaphone may be providing a CDN.

pbatard, 2025-07-31:
> Filename or an internal identifier?

Filename.

> Has the filename changed to ie `windows disk maker.exe`?
> 
> Or does Fort Filewall change the identifier to ie `suspicious application named rufus.exe`?
> 
> Are those values logged?

I'm sorry but I have no idea what you're on about. As indicated in the FAQ entry, when Rufus detects that its filename is `rufus.exe` it _should_ (haven't tested it in a long time) bypass the prompt that asked whether you want to enable the check for update, and consider that the user wants the check for update enabled.

You're going to have to elaborate on **why** you are suddenly asking these questions, that are logically completely unrelated to the behaviour described in the FAQ entry that's explicitly linked before the part you chose to quote...

pbatard, 2025-08-02:
Since I made a note to **potentially** look for alternate ways to get to the issuer certificate without invoking `CertGetCertificateChain()` (but again, this will be **low priority stuff**), I'm going to close this issue.

closed 2025-08-02 by pbatard