pbatard/rufus #2788: Feature Request: add option to create Vanilla Windows Setup Disk without any SKIPPING steps

evan0greenup, 2025-08-14:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

This is not bug so the below steps are skipped
Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Even if I disable all the custom options of Rufus for simplifying installations steps, the created boot-able USB disk is still not vanilla.

When boot from the created disk, it will directly enter the Hard disk selection step, all the previous steps (e.g. "Single Button `Start Installation`"; "Enter Product Key/ Checkbox of I don't have product key", ...) is not shown.

It would be nice to add a option in Rufus to just simply create Vanilla Windows Setup USB Drive without any additional modification. Just have the same outcome as Microsoft Media Creation Tool.

Log
---
```
Rufus x64 v4.9.2256
Windows version: Windows 11 Pro x64 (Build 26100.4946)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Linux File-Stor Gadget USB Device' (1D6B:0104)
Device eliminated because it appears to contain no media
0 devices found
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
UEFI bootloaders analysis:
  â€¢ /bootmgr.efi*
  Signed by 'Microsoft Windows Production PCA 2011'
  Note: This bootloader may fail Secure Boot validation on systems that
  have been updated to use the 'Windows UEFI CA 2023' certificate.
  â€¢ /efi/boot/bootx64.efi*
  Signed by 'Microsoft Windows Production PCA 2011'
  Note: This bootloader may fail Secure Boot validation on systems that
  have been updated to use the 'Windows UEFI CA 2023' certificate.
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 26100.1742)
  Size: 5.4 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.1.13)
Using image: Win11_24H2_English_x64.iso (5.4 GB)
Checking for UEFI DBX updates...
Checking for Rufus updates...
Checking release channel...
No new release version found.
```

pbatard, 2025-08-14:
> the created boot-able USB disk is still not vanilla

That's where you are misunderstanding what "vanilla" is.

Nothing is vanilla when it comes to ISO &rarr; USB conversion, so the drive created by the MCT is **not** vanilla either. Nothing is.

However, just like Rufus, the MCT can guarantee that the all the individual files from the ISO are present, and that they are 100% identical. **BOTH** Rufus and MCT provide that (if you uncheck the WUE options). Contrary to what you are asserting, and which I can guarantee you have not thoroughly researched (because it's easy to check if Rufus actually modifies files by comparing their hashes against the ones from the ISO, a process that can easily be automated and accomplished), if you don't use any WUE options, then Rufus does add "any additional modification".

Or perhaps your interpretation of "vanilla" is _"whatever the MCT does"_, but in that case, you might as well complain that Rufus will not use the same IDs for the partition(s) it creates on the media as the MCT, or the multiple **utterly irrelevant details that have absolutely no bearing on how the final installation of Windows will behave**, such as the creation of an extra partition (which is just there for boot, and which, again, will be completely **ignored** by the installation process once it starts).

Please don't blindly trust what you read on the internet about _"Rufus modifying Windows files"_, because that is completely and utterly wrong. Instead, do your own research, and only log an issue if you believe that you have evidence (which again, I can 100% guarantee you will not be able to find) that, using the same source ISO (be weary that if you do not use an ISO as source, the MCT will fetch Windows files on the fly, that are different from the ones found in the latest release ISO) an installation of Windows performed by a media created by the MCT and an installation of Windows performed by a media created by Rufus will perform **IN EXACTLY THE SAME WAY**. And that is because, as you can again very easily validate, you will find that **all** the installation files from the source ISO were exactly the same.

As a result, I will close this issue.

closed 2025-08-14 by pbatard

pbatard, 2025-08-19:
> Even if I uncheck all WUE option, the boot device created by Rufus is still skip some steps like `Click Start Install Button`, `Agree Agreement`, `Input product key`, ...

I am not seeing any of that **in actual tests with *vanilla* Windows ISOs** (such as the Windows 11 24H2 retail US ISO downloaded from https://www.microsoft.com/en-US/software-download/windows11) on real hardware. I got the `Agree Agreement`, `Input product key` and so on with all the WUE options unchecked, so I can 100% invalidate your statement that these will be somehow skipped if you create the media by Rufus. I have validated that, contrary to what you are stating, these were appearing as expected.

And again, since Rufus does not alter the Windows ISO content if you uncheck the WUE options, I can guarantee that, when you don't have any WUE option selected, the media created by Rufus will behave in exactly the same way as if you had not used Rufus to create it, i.e., that you are not going to get any more or any less options and prompts.

I will therefore have to ask you to formally validate this, rather than report on what you **think** is happening, and be **very** mindful that, depending on where you get your media, and whether it was altered, such as by adding an `ei.cfg` file, the setup behaviour will be different. However, this difference in behaviour will be 100% due to your boot media, and not due to Rufus.