pbatard/rufus #2794: Rufus closes after I hit start on Windows 11 24H2

luluhoc, 2025-09-02:
Checklist
---------
- [x ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Rufus closes after I hit start on Windows 11 24H2. I have tried multiple versions of Rufus, including the portable ones. I also tested different USB sticks and ports, and even formatted the USB manually, but nothing seems to work. I even tried creating a non-bootable USB. The issue seems to be with the formatting as soon as Rufus tries to format, it crashes. I have also disabled windows defender, I don't have any antiviruses installed.

Log
---
```
[2572] *** Rufus init *** 
[2572] Cur dir: 'C:\Users\lucja\Desktop\' 
[2572] App dir: 'C:\Users\lucja\Desktop\' 
[2572] Sys dir: 'C:\Windows\system32' 
[2572] Usr dir: 'C:\Users\lucja' 
[2572] Dat dir: 'C:\Users\lucja\AppData\Local' 
[2572] Tmp dir: 'C:\Users\lucja\AppData\Local\Temp\' 
[2572] Binary executable is signed by 'Akeo Consulting' 
[2572] Will use settings from INI file 
[2572] found locale 'en-US' 
[2572] loc file not found in current directory - embedded one will be used 
[2572] localization: extracted data to 'C:\Users\lucja\AppData\Local\Temp\Ruf598C.tmp' 
[2572] localization: found locale 'en-US' 
[2572] localization: found locale 'ar-SA' 
[2572] localization: found locale 'bg-BG' 
[2572] embedded.loc(1040): the version of this translation is older than the base one and may result in some messages not being properly translated.
[2572] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[2572] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[2572] localization: found locale 'zh-CN' 
[2572] localization: found locale 'zh-TW' 
[2572] localization: found locale 'hr-HR' 
[2572] localization: found locale 'cs-CZ' 
[2572] localization: found locale 'da-DK' 
[2572] localization: found locale 'nl-NL' 
[2572] localization: found locale 'fi-FI' 
[2572] localization: found locale 'fr-FR' 
[2572] localization: found locale 'de-DE' 
[2572] localization: found locale 'el-GR' 
[2572] localization: found locale 'he-IL' 
[2572] localization: found locale 'hu-HU' 
[2572] localization: found locale 'id-ID' 
[2572] embedded.loc(6250): the version of this translation is older than the base one and may result in some messages not being properly translated.
[2572] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[2572] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[2572] localization: found locale 'it-IT' 
[2572] localization: found locale 'ja-JP' 
[2572] localization: found locale 'ko-KR' 
[2572] localization: found locale 'lv-LV' 
[2572] localization: found locale 'lt-LT' 
[2572] embedded.loc(8235): the version of this translation is older than the base one and may result in some messages not being properly translated.
[2572] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[2572] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[2572] localization: found locale 'ms-MY' 
[2572] embedded.loc(8625): the version of this translation is older than the base one and may result in some messages not being properly translated.
[2572] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[2572] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[... 194 lines cut ...]
[33500]   Name: 'HFSPLUS' 
[33500]   Detected File System: HFS/HFS+ 
[33500]   ID: {690055A7-4F89-4255-8877-AD68D9FCB7C7} 
[33500]   Size: 1.8 GB (1982582784 bytes) 
[33500]   Start Sector: 10528, Attributes: 0x1000000000000001 
[33500] Partition 4: 
[33500]   Type: Microsoft Basic Data Partition 
[33500]   Name: 'Gap1' 
[33500]   Detected File System: (Unrecognized) 
[33500]   ID: {690055A7-4F89-4255-8870-AD68D9FCB7C7} 
[33500]   Size: 300 KB (307200 bytes) 
[33500]   Start Sector: 3882760, Attributes: 0x1000000000000001 
[33500] Ignoring drive '\\.\H:' as it spans multiple disks (RAID?) 
[33500] Found 4 active Secure Boot certificate entries from remote 
[33500] Found 1 revoked Secure Boot certificate entries from remote 
[33500] Ignoring volume '\\?\Volume{b56b6d43-0000-0000-0000-605a03000000}' because it has more than one extent (RAID?)... 
[33500]  
[33500] Format operation started 
[33500] Requesting disk access... 
[33500] Ignoring drive '\\.\H:' as it spans multiple disks (RAID?) 
[33500] Ignoring drive '\\.\H:' as it spans multiple disks (RAID?) 
[33500] No drive letter was assigned... 
[33500] Will use 'F:' as volume mountpoint 

```

pbatard, 2025-09-02:
I guess you were trying to format a drive that didn't really exist then. I'm not sure how one can have a disk that's disconnected and yet still present (and treated like an active disk). Can you explain how you accomplished that? Did you map the drive to a letter or something, and then removed the disk?

Also, you might want to try switching to VDS for formatting to see if that helps when the issue occurs. In Rufus press <kbd>Alt</kbd>-<kbd>V</kbd> and you should see `VDS enabled` on the Rufus status bar.

luluhoc, 2025-09-03:
No, not at all. I was trying to format an existing drive. It's just that when there was a disk with a letter assigned to it, I had disconnected it, so it was showing with a red X icon in Disk Management.

pbatard, 2025-09-03:
I'm afraid I still don't understand your situation.

> I was trying to format an existing drive.

So you weren't trying to format the drive that got disconnected?

>  It's just that when there was a disk with a letter assigned to it,

Does 'it' refer to the drive you were trying to format? Or some other separate disk. This statement is very confusing to me, because I have no idea what you are referring to here.

> I had physically disconnected it, so it was showing with a red X icon in Disk Management.

So you had some kind of disk `X:`, that had a letter assigned to, that you disconnected, that was different from the physical disk you were trying to format/partition in Rufus (??), but that was still showing in Disk Management after you ejected it? How is that possible, when, if you properly eject a disk, then Disk Management should reflect the disk having been removed, regardless of whether a letter was assigned or not.

Again, if you want me to do anything about this, then you need to provide more information on how one is supposed to replicate your situation, because, right now, I have no idea how to make a mapped drive appear with a red X icon in disk management, without the disk being defective (because in most situations, even if you forcefully yank a drive, Windows should just reflect that it is no longer present.

How do you get into a situation where, as you stated, "disconnected disks are not removed from Disk Management"? What kind of disk connection did it have (SATA/SSD/USB)? And what did you do exactly to "disconnect" that disk?

jatinhbhatti, 2025-09-07:
Hello @luluhoc ,
I faced the same issue, but I solved it by running Windows in Safe Mode and creating a bootable pen drive from there. I hope this works for you too

pbatard, 2025-09-07:
@jatinhbhatti, considering that you wrote the same comment for two completely different errors (see #2795), I strongly doubt that you got the two exact completely different issues (one where Rufus crashes, and one where it merely reports "undertermined error" without crashing).

Please avoid commenting unless you are 100% sure that you ran into the same issue (in which case, it would really help to post your full log, so that we can validate and find relevant elements).

jatinhbhatti, 2025-09-07:
Let’s connect I will explain exactly what happened in this issue

Again, if you want me to do anything about this, then you need to provide more information on how one is supposed to replicate your situation, because, right now, I have no idea how to make a mapped drive appear with a red X icon in disk management, without the disk being defective (because in most situations, even if you forcefully yank a drive, Windows should just reflect that it is no longer present rather than show it with an error).

So you had some kind of disk X:, that had a letter assigned to, that you disconnected, that was different from the physical disk you were trying to format/partition in Rufus (??) yes, but that was still showing in Disk Management after you ejected it? yes

luluhoc, 2025-09-07:
> Hello [@luluhoc](https://github.com/luluhoc) , I faced the same issue, but I solved it by running Windows in Safe Mode and creating a bootable pen drive from there. I hope this works for you too

Thanks, although I deleted the disconnected disks from Disk Management and it resolved the problem

pbatard, 2025-09-07:
@jatinhbhatti, I call bullshit. You can not have two completely different errors from two completely independent issues and claim they're the same. And you are disrupting this thread (this is not a call centre where you get dedicated face to face support) so I am banning you and deleting your comments.

@luluhoc,

You still haven't answered **how** you disconnected the drive. Did you go through safe removal or yanked it. Or did you have some application that was still accessing the drive at that time and you used force removal. Again, I cannot do anything about this situation unless you describe in details how one can get into a situation where a drive can be removed but still appear in disk manager.

The thing is Windows *automatically* assigns a letter to any drive with a partition you can mount, so you don't have to assign one manually. And when you go through safe removal, that letter is deleted and removed from Disk Management.

And even if you do something like opening a dos prompt on the drive (so that safe removal complains that it cannot eject the disk because it is in use) and yank the drive, I still see the drive disappearing from Disk Manager.

Therefore, there is something more custom in what you are doing, that you have not explained, and that prevents me from investigating this issue any further. You are doing something "extra" to get into a situation where Disk Manager shows a phantom, drive still being attached, and I am really hoping that you can figure out (and described) exactly what it is, because, unless you can do that, there's really nothing I can do here.

luluhoc, 2025-09-07:
@pbatard I really don't remember, and I'm not entirely sure how it came to be that the disconnected disk was showing as failed...

I just disconnected when the PC was off...

I have found this issue 
https://www.reddit.com/r/Windows10/comments/hvpnq0/missing_device_shows_as_failed_in_disk_management/

pbatard, 2025-09-07:
> I just disconnected when the PC was off...

Now we are getting somewhere. Disconnecting a drive while you put your PC in sleep mode is quite important information.

And yes, I have no doubt that you and other people got into the situation where you get a "Missing" disk being reported in Windows. I am not disputing the veracity of your claim. But I want to know is how the heck one can get into such a situation. Have you also been using dynamic disks like the link you pointed to then?

luluhoc, 2025-09-07:
> Now we are getting somewhere. Disconnecting a drive while you put your PC in sleep mode is quite important information.
>
> And yes, I have no doubt that you and other people got into the situation [...]

The PC was off not in a sleep mode, I can confirm that, yeah it was as dynamic disk.

amoki455, 2025-10-01:
I managed to replicate the issue in a VM with VMware; it appears unrelated to Windows version because my host is Windows 11 and the VM is Windows 10, and both encountered the issue.

### We can create a “Missing (Failed)” disk in a VMware VM as follows:

**1. Add two hard disks to the VM through Virtual Machine Settings.**

[image]

**2. Create a new spanned volume using the two disks (in my case Disk 0 and Disk 3).**

[image]

[image]

**3. Open Virtual Machine Settings and remove one of the two disks; this will cause a missing disk to appear in Disk Management.**

[image]

### For the USB drive, you can use a physical one or create a virtual USB disk as follows:

1. cd "C:\Program Files (x86)\VMware\VMware Workstation"
2. vmware-vdiskmanager.exe -c -a buslogic -s 32GB -t 0 "D:\VMware VMs\vm-usb-key.vmdk"
3. Ensure these lines are in your VM's .vmx file:
```
ehci.present = "TRUE"
ehci.pciSlotNumber = "xx"
```
4. Add these lines to your VMX file
```
ehci:0.present = "TRUE"
ehci:0.deviceType = "disk"
ehci:0.fileName = "vm-usb-key.vmdk"
ehci:0.readonly = "FALSE"
```
5. Save the .vmx file, double-click it to open the VM, start the VM, and format the virtual USB drive.

### Debug steps:

1. Copy the VS Remote Debugger from: C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\Remote Debugger to the VM and run msvsmon.exe.
2. Create a debug build, copy it to the VM, and run Rufus.exe.
3. On the host, choose Attach to Process from the Debug menu in Visual Studio.
4. Enter the VM's remote debugger address and select the Rufus process.
5. You don't need to specify an ISO — choose Non bootable.
6. Click Start.

### Results:

- **This is where the exception occurs.**

[image]

- **This is the call stack; the last call from Rufus code appears to be `GetVdsDiskInterface`.**

[image]

- **It looks like the issue here is that `prop.pwszName` is `NULL`.**

[image]

pbatard, 2025-10-01:
@amoki455, thank you so much for spending time replicating the issue and debugging it!

What you found is definitely the problem, and it should be an easy fix.

Once I push the fix, it will close this issue, but I will point to a build that can be used to test the fix. If by any chance you still see an application crash in the same circumstances, feel free to let me know.

I will also see if I can recreate a VMWare environment to also test this, but it might take a while.

At any rate, you really have my thanks for figuring out this hard to troubleshoot issue!

pbatard, 2025-10-01:
NB: You can find the test builds with the fix in the VS2025 Artifacts of https://github.com/pbatard/rufus/actions/runs/18159619272.

commit 2a8c066b (2025-10-01): [core] fix an application crash when the platform has a missing/failed dynamic disk

* With massive thanks to @amoki455 for helping figure that one out!
* Closes #2794.
* Also make sure we release pDisk on all conditions.