pbatard/rufus #2797: New ability to manually add storage exclusion lists

lala57188, 2025-09-07:
[x] When I create a BiliS OS boot disk, the application may mistakenly select a large-capacity USB drive when formatting the boot disk, thereby accidentally deleting important data. I hope to add a new function to manually add an exclusion list to prevent accidental deletion of files.

pbatard, 2025-09-07:
1. Check list is not optional.
2. Log is not optional. Clearly, if as you allege, Rufus selects a large capacity drive instead of a smaller one, you should be able to replicate the problem and provide a log.
3. Rufus is actually designed to list disks starting from the smallest to the largest, and also not list large drives by default, which is why, again, because you chose not to provide a log or any relevant data, I don't see anything for me to do, and I will close this issue.

pbatard, 2025-09-07:
Also, if you had read the FAQ, as you were explicitly asked to do in the check list you ignored, you would have found:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_can_I_prevent_a_specific_drive_from_being_listed_by_Rufus

closed 2025-09-07 by pbatard