pbatard/rufus #2798: Rufus 4.9.2256 is not able make successfully an .vhdx image of a 64gb pendrive with Ubuntu 22.04

gfernval, 2025-09-07:
Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

gfernval, 2025-09-07:
I made an Ubuntu 22.04 bootable USB pendrive (Installing Ubuntu on a 64gb Kingston pendrive), booting such USB pendrive works fine. Then, using Windows 10.0.19045.2965 and Rufus 4.9.2256 I make an image of the 64gb pendrive by clicking the floppy icon located at the right of "Device" dialog box (appears "Save As"...NO_LABEL.vhdx)
Then, when I try to copy the .vhdx file generated back to other identical 64gb pendrive gives an error about an "image too large". I patched the source code to disable such error, Rufus then is able to copy the .vhdx back to a new identical 64gb USB pendrive but this pendrive is unable to boot Ubuntu 22.04
It seems to me that rufus is not able to copy a pendrive containig more that 1 partition. The USB pendrive containing Ubuntu 22.04 contains 3 partitions: UEFI, swap, and filesystem

pbatard, 2025-09-07:
Also, I just tested and saw no issue saving a drive to VHDX and then writing the same image to the same drive (without seeing any complaint about the size being too small), so I am pretty confident, since you state **"to other identical 64gb pendrive"** that you fail to realize that 64GB for one pendrive is not 64GB for another. For instance, if your second pendrive is smaller by even one sector, it will still report to be 64GB, but the backup GPT will then be screwed up.

In the absence of logs, the fact that Rufus did report that your drive was too small makes me pretty confident that the drive you wrote the image to was indeed too small, and, since you chose not to provide a log corroborate or invalidate that case, I will close this issue.

closed 2025-09-07 by pbatard