pbatard/rufus #2807: Rufus doesn't add persistent flag to "label live" in isolinux/live.cfg when persistent option is enabled

yulian5, 2025-09-19:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I flashed latest Linux Mint 22.2 Cinnamon with persistent option 10gb and then noticed that **persistence actually doesn't work** when booting using BIOS: I installed VeraCrypt which was missing after reboot.
After some investigation I found that my old I PC boots in only BIOS and live entry in **isolinux/live.cfg**  have **persistent** flag missing. After I added it, persistence starts to work:

label live
	menu label Start Linux Mint
	menu default
	kernel /casper/vmlinuz
	append boot=casper **persistent** initrd=/casper/initrd.lz uuid=6e72f523-dc09-4880-8910-93ffa64401c5 username=mint hostname=mint quiet splash --

But other entries in **isolinux/live.cfg** have persistent flag.

One more problem: casper-rw is formatted as **ext3**, I guess it should be **ext4**. I reformatted my casper-rw as ext4.

Thanks!

Log
---
```
<FULL LOG>
```

pbatard, 2025-09-19:
Thanks for the report.

I'll see if I can improve the persistent option for Mint BIOS users, but considering that fewer and fewer people are going to be using Mint on BIOS/Legacy system, and that trying to support persistence for every distro under the sun [is a major pain in the ass](https://github.com/pbatard/rufus/blob/329e9705e2b3d8220ad88e73aac5e7f5b9c7e4b3/src/iso.c#L483-L520) because, since persistence is implemented in a **very idiotic way** on Linux, we have no choice but use a super brittle mechanism in Rufus, and the minute distro maintainers start to move their kernel options around, it's going to break.

> One more problem: casper-rw is formatted as ext3, I guess it should be ext4.

If it has to be `ext4` for persistence to work on some Linux distros, then persistence implementations on Linux would have to be even stupider than they currently are, which is to say something (again, **if** Linux folks were doing persistence properly, there should be **no need** for a kernel options or a specific file system, but just for the init script to look for a mountable partition with a specific label/file/UUID and that's it. It's **really** not that complicated!), so I doubt that.

`ext3` is absolutely fine for a persistence partition running off a USB drive, and, since [I have to provide full file system support for a file system that is not native to Windows](https://github.com/pbatard/rufus/tree/master/src/ext2fs), just to implement persistence support in Rufus (and this was a lot more involved than simply copying a bunch of existing source files, as ext2fs was never designed to compile for Visual Studio/Windows), I have no plan to provide `ext4` support in Rufus, as the cost/benefit ratio of doing so is simply not worth it.

yulian5, 2025-09-19:
> replacing kernel with linux would immediately break Ubuntu, Debian and so on

I was not sure do you download live.cfg or generate it. But menu entries in the same config have both: `kernel` and `linux`.

commit 1c201cc0 (2025-09-24): Rufus 4.10 (Build 2279)

* Improve persistence support for Linux Mint.
* Closes #2807.
* Also fix the Swedish translation.
closed 2025-09-24 by pbatard